ome/bioformats #2869: Update from: "bfconvert: fix plane indexes for multi-series exports #1839" .vsi files having similar problem

AlexBarrington, 2017-06-13:
I am using the example dataset provided at http://biop.epfl.ch/TOOL_VSI_Reader.html
I am attempting to convert their "Test.vsi" to all tiled tiff images with this code: 

**bfconvert -tilex 512 -tiley 512 ~/VSI_Sample_Dataset/Test.vsi ~/Example_tifs/example_%x_%y_%m.tif** 

From this code, I get this as output: 

**CellSensReader initializing ~/VSI_Sample_Dataset/Test.vsi
[CellSens VSI] -> ~/Example_tifs/example_%x_%y_%m.tif [Tagged Image File Format]
Tile size = 512 x 512
Switching to BigTIFF (by file size)
	Series 0: converted 1/1 planes (100%)
Tile size = 512 x 512
Exception in thread "main" loci.formats.FormatException: Plane index:1 must be < 1
	at loci.formats.FormatWriter.checkParams(FormatWriter.java:387)
	at loci.formats.out.TiffWriter.saveBytes(TiffWriter.java:215)
	at loci.formats.tools.ImageConverter.convertTilePlane(ImageConverter.java:748)
	at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:634)
	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:563)
	at loci.formats.tools.ImageConverter.main(ImageConverter.java:884)**

I also tried this with "-series 1" added and got this as output: 

**CellSensReader initializing ~/VSI_Sample_Dataset/Test.vsi
[CellSens VSI] -> ~/Example_tifs/example_%x_%y_%m.tif [Tagged Image File Format]
Tile size = 512 x 512
Exception in thread "main" java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:653)
	at java.util.ArrayList.get(ArrayList.java:429)
	at ome.xml.model.Pixels.getChannel(Pixels.java:628)
	at ome.xml.meta.OMEXMLMetadataImpl.getChannelSamplesPerPixel(OMEXMLMetadataImpl.java:1232)
	at loci.formats.FormatWriter.getSamplesPerPixel(FormatWriter.java:469)
	at loci.formats.out.TiffWriter.getPlaneCount(TiffWriter.java:451)
	at loci.formats.FormatWriter.setId(FormatWriter.java:352)
	at loci.formats.out.TiffWriter.setId(TiffWriter.java:154)
	at loci.formats.ImageWriter.setId(ImageWriter.java:481)
	at loci.formats.tools.ImageConverter.convertTilePlane(ImageConverter.java:718)
	at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:634)
	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:563)
	at loci.formats.tools.ImageConverter.main(ImageConverter.java:884)**

The version I am using: 

**bfconvert -version
Version: 5.5.1
Build date: 25 May 2017
VCS revision: fcd74073cbb30e9e9a7618fbc30ee310dbc2357c**

I thought this had been resolved in "bfconvert: fix plane indexes for multi-series exports #1839", but I am coming across the same issue in 5.5.1. Is this again a simple fix like it was with #1839? Any help would be greatly appreciated.

dgault, 2017-06-14:
Hi @abarrington, thank you for reporting this. Using the same sample file I was able to reproduce the same behaviour that you were seeing. The issue appears to occur when using "_%x_%y_%m" to save the tiles individually.

The first exception does look very similar to https://github.com/openmicroscopy/bioformats/pull/1839 except its only occurring this time when writing individual tiles. Tested using "_%s_%z_%t" appears to be working correctly.

The second exception is actually similar to another report we received recently https://github.com/openmicroscopy/bioformats/issues/2847 which was fixed in https://github.com/openmicroscopy/bioformats/pull/2848. 

Hopefully similar fixes to the two previous solutions can be put in place for the use of "_%x_%y_%m", it would also be good if we could have additional testing for these options to catch this in the futire.

abarrington, 2017-06-14:
Hi @dgault, thank you. That fix worked.

abarrington, 2017-06-16:
Hi @dgault, while that fix did work, I have not been able to open the files. From my Mac, Preview is saying that it does not recognize the file format. Is it possible that the .vsi to .tif conversion is not converting correctly?

abarrington, 2017-06-29:
Hi @dgault, I have come across a similar issue, but cannot determine the cause. 

From the following line:
**bfconvert -bigtiff -series 7 /path/test.vsi ~/test.tiff** 

I get: 
**/path/test.vsi
CellSensReader initializing /path/test.vsi
[CellSens VSI] -> ~/test.tiff [Tagged Image File Format]
Tile size = 512 x 512
Exception in thread "main" java.lang.NullPointerException
	at java.lang.System.arraycopy(Native Method)
	at loci.formats.in.CellSensReader.openBytes(CellSensReader.java:538)
	at loci.formats.FormatReader.openBytes(FormatReader.java:886)
	at loci.formats.ImageReader.openBytes(ImageReader.java:441)
	at loci.formats.tools.ImageConverter.convertTilePlane(ImageConverter.java:698)
	at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:634)
	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:563)
	at loci.formats.tools.ImageConverter.main(ImageConverter.java:884)**

I have determined that series 7 does in fact exist, and it is the desired series to extract from the entire file. 

I also get this error with showinf:

**Reading series #0 pixel data (0-0)
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
	at loci.common.DataTools.allocate(DataTools.java:923)
	at loci.formats.FormatReader.openBytes(FormatReader.java:877)
	at loci.formats.ImageReader.openBytes(ImageReader.java:441)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
	at loci.formats.gui.BufferedImageReader.openImage(BufferedImageReader.java:86)
	at loci.formats.tools.ImageInfo.readPixels(ImageInfo.java:818)
	at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1051)
	at loci.formats.tools.ImageInfo.main(ImageInfo.java:1117)**

This happens after showing the metadata of each series. 

Thank you in advance.

dgault, 2017-06-30:
Hi @abarrington,

This looks to be very similar to another reported issue https://trello.com/c/fUhH0Cxo/141-vsi-npe-in-openbytes

I have spent some time today debugging the original report and it appears to have been the result of an unknown compression type.

Do you know what compression type is being used for each of the series in your image?
Currently the reader has support for Raw, JPEG, JPEG 2000, PNG and BMP

abarrington, 2017-06-30:
Hi @dgault,

The series are Raw compressed.

abarrington, 2017-07-20:
Hi @dgault, do you have any updates on this issue? Thanks!

dgault, 2017-07-20:
Hi @abarrington,

Without any official documentation listing this unknown compression types I have been testing through trial and error.  From the original sample file we received with the previous report of this incident it appears that in that particular instance a JPEG encoding was being used.

As the sample file you were using was Raw compressed it would be good to confirm the particular encoding. Would you have a small sample file showing the issue that could be uploaded to https://www.openmicroscopy.org/qa2/qa/upload/?

abarrington, 2017-07-20:
Hi @dgault,

I will work on getting something uploaded. In the mean time, I have been attempting to use the MATLAB toolbox to open the files. I keep getting the following error:
[image]

Is there a way to import chunks of the file at a time and then restitch them together afterward using these tools? That would avoid the large file size and Java not liking it.

sbesson, 2017-07-20:
@abarrington: `bfGetPlane(r, iPlane);` returns the whole plane independently of the image dimensions which indeed is problematic for large images. You can load individual tiles using `bfGetPlane(r, iPlane, x, y, w, h)` where `(x, y)` and `(w, h)` specify the origin and the dimension of the tile to return respectively.

abarrington, 2017-07-24:
@sbesson: I tried this method, thank you for the suggestion. It seemed to work, however, I still tend to get images with most of the data as black in the image.

abarrington, 2017-07-26:
@sbesson: As an update, I have been able to get these converted slide by slide. However, I noticed that when I use parfor to batch process my files, it tends to yield a NullPointerException. It seems that the reader variable from bfGetReader is not very compatible passed across cores. I am not sure if there is a way to improve this kind of issue, but it would be useful to have a better way to batch process these.

abarrington, 2017-08-01:
@sbesson: This seems to work for the majority of my files, but I tend to get this for some of my .vsi files: 
[image]

dgault, 2017-08-02:
Hi @abarrington, is it constantly failing on the same images? 
Does varying the requested tile size have any effect on the error occurring?

abarrington, 2017-08-02:
Hi @dgault, yes, it fails on the same images. No, it does not appear to affected the error. I'm trying again without parallelizing, which might be causing some of the issue. For some of the slides that produce the error during batch conversion, I am able to get a single plane at a time. However, I cannot extract anything from a handful of the .vsi files because of the null pointer exception. It's possible that MATLAB's parfor loops do not play nice with the Java code implemented. I personally do not know enough of how that would work, but that might be where it is faulting. I will update after the batch finishes without parallelizing. Let me know if I can provide anything else useful (with the exception of the actual .vsi files, which I am not able to provide).

sbesson, 2017-08-07:
@abarrington: answering the parallelization question, the Bio-Formats API and especially readers should not be treated as thread-safe. To work multi-threader, the current approach is to initialize one reader per thread. When the initialization time is large, it can be reduced by pre-initializing and caching the initialized reader using the `Memoizer` reader wrapper - see docs.openmicroscopy.org/latest/bio-formats5/developers/matlab-dev.html#improving-reading-performance for an example of usage of `parfor` with multiple reader initializations.

abarrington, 2017-08-07:
@sbesson: thank you! That helps a lot.

sbesson, 2017-08-10:
@abarrington glad it helped. Closing this issue assuming the NPE you received will be resolved by initializing per-thread readers as suggested in https://github.com/openmicroscopy/bioformats/issues/2869#issuecomment-320713231. Keep us posted if you have additional concerns either as a comment on this issue or in a new one.

closed 2017-08-10 by sbesson