ome/bioformats #2898: BioFormats 5.5.3 does not read CZI metainfo (series number, pyramid levels) correctly anymore

sebi06, 2017-07-17:
Hi guys,

I just tries to use Grid-Stitching in Fiji to stitch the CZI (see link) to check if this plugin can deal with BioFormats feature to read CZI pyramid levels.
 It did not work because the plugin claimed this file has only 5 series. But it is a 15x8 TileRegion.

https://www.dropbox.com/s/kytfnxf2dpywzf7/15x8_Tileregion.czi?dl=0

Error Message:
ERROR: Inconsistent series layout. Metadata says 5 tiles, but contains only 120 images/tiles.
ERROR: Error during tile discovery, or invalid grid type. Aborting.

[image]

I tested this using BioFormats 5.1.10 and got what I expected 120 Tiles.

[image]

When I do the same using BioFormats 5.5.3 i get only 5 series, which of course is not correct. So something went wrong when reading the data. This image has of course some pyramid levels, but at full resolution there are 120 tiles. Obviously this beraks a lot of existing workflows if that is really true ...

[image]

And it raises the additional question, if the Grid-Stitching plugin makes use of the Bioformats option to only read the full-res level of the pyramid.

sbesson, 2017-07-17:
Hi @sebi06, you are correct that the total number and dimensions of the series is now dependent on the reader options while in Bio-Formats 5.1.10, the only available possibility was to read individual tiles. Using the sample file you provided and varying the [Zeiss CZI options](http://docs.openmicroscopy.org/bio-formats/5.5.3/formats/options.html), I have the following results:

```
sbesson@ls30630:Downloads $ showinf -nopix -nometa 15x8_Tileregion.czi | grep Series\ c
Series count = 5
sbesson@ls30630:Downloads $ showinf -nopix -nometa -option zeissczi.autostitch false 15x8_Tileregion.czi | grep Series\ c
Series count = 121
sbesson@ls30630:Downloads $ showinf -nopix -nometa -option zeissczi.autostitch false -option zeissczi.attachments false 15x8_Tileregion.czi | grep Series\ c
Series count = 120
```

For the Grid-Stitching plugin, as the code is using [BF.openImagePlus()](https://github.com/fiji/Stitching/blob/Stitching_-3.1.3/src/main/java/plugin/Stitching_Grid.java#L840) for initializing the reader, I would expect that unchecking the `Automatically stitch tiled images` and `Include attachment images` checkboxes for the Zeiss CZI format in the Configuration window should be sufficient to allow the plugin to read the data in the expected layout. Could you check this? /cc @StephanPreibisch

sebi06, 2017-07-18:
This is what I already tried yesterday, but it does not work.

[image]

But I think the Grid Stitching is just no aware of this since it counts the image attachment and the actual 4 pyramid levels while BioFormats 5.5.3 actually correctly got 4 real series.

[image]

sbesson, 2017-07-18:
Thanks for checking and for the debugging screenshots. I think they help identifying the source of the problem.

The error message informs about a failing [check](https://github.com/fiji/Stitching/blob/Stitching_-3.1.3/src/main/java/plugin/Stitching_Grid.java#L1006) between the size of two lists:

- `imps` is an array of `ImagePlus` objects created using `BF.openImagePlus()` which internally consumes the options defined in the ImageJ configuration window
- `elements` is ArrayList of ImageCollectionElement objects created in another call to the low-level Bio-Formats API notably and notably the [creation of a separate reader](https://github.com/fiji/Stitching/blob/master/src/main/java/plugin/Stitching_Grid.java#L863)

The two reader stacks are created independently and in the context of this issue, the main problem is that they have mismatching options. There are a few ways here to tackle this issue in the Grid Stitching plugin:

- the immediate workaround is be to copy the logic in [LociPrefs](https://github.com/openmicroscopy/bioformats/blob/v5.5.3/components/bio-formats-plugins/src/loci/plugins/util/LociPrefs.java#L96)  and set the same metadata options
- it might be possible to introspect the options of the reader created by `BF.openImagePlus()` but that would require a bit more reorganization in the Grid Stitching code
- there might be case for a new API that would either return the `MetadataOptions` defined by the Configuration window or apply them to a reader although this would require more discussion and scoping.

zeissmicroscopy, 2017-07-22:
Hi Sebastian,

if I understood you correctly this workaround would be to modify the BioFormats sourceCode directly, right? But I would be better, if the Grid Stitching plugin could be modified so that it works with the latest BioFormats and CZI in my opinion. So maybe I should file a request for that (but where? Bug Entry)?

And the workaround does not really solve the issues that users currently using CZIs and GridStitching are still stuck. 

What about downgrading to Bioformats 5.1.10 as the last (emergency) option?

imagejan, 2017-07-22:
@zeissmicroscopy wrote:
> So maybe I should file a request for that (but where? Bug Entry)?

PRs are accetped at https://github.com/fiji/Stitching/pulls, and issues can be filed here: https://github.com/fiji/Stitching/issues

sbesson, 2017-08-08:
@ zeissmicroscopy: for this particular issue, the fix has to be applied at the Grid Stitching plugin source code level in order to ensure that the two readers are created with the same options - ideally the ones set up by the user in the Configuration window. In that sense, I am closing this issue in favor of https://github.com/fiji/Stitching/issues/42 that you opened and we will start watching it for any API discussion.

closed 2017-08-08 by sbesson