ome/bioformats #2919: BioFormats Error when CZI file in loci.common.DataTools

sebi06, 2017-08-07:
Hi guys,

on Ubuntu 17.04 (Virtual Machine) i tried to open the following CZI via BioFormats Importer:

https://www.dropbox.com/s/zxtbxfhsnc7vf1o/OverViewScan_Test_raw_stitched.czi?dl=0

and run into serious errors. The Fiji is up-to-date. Without really having a great clue, I renamed **formats-common-5.1.10.jar** to **formats-common-5.1.10.jar.old** and then it seems to work fine ... (???)

Errors:

```
WARNING: multiple locations found!
java.lang.NoSuchMethodError: loci.common.DataTools.parseDouble(Ljava/lang/String;)Ljava/lang/Double;
	at loci.formats.in.ZeissCZIReader.translateInformation(ZeissCZIReader.java:2061)
	at loci.formats.in.ZeissCZIReader.translateMetadata(ZeissCZIReader.java:1783)
	at loci.formats.in.ZeissCZIReader.initFile(ZeissCZIReader.java:1028)
	at loci.formats.FormatReader.setId(FormatReader.java:1397)
	at loci.plugins.in.ImportProcess.initializeFile(ImportProcess.java:498)
	at loci.plugins.in.ImportProcess.execute(ImportProcess.java:141)
	at loci.plugins.in.Importer.showDialogs(Importer.java:140)
	at loci.plugins.in.Importer.run(Importer.java:76)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:217)
	at ij.IJ.runPlugIn(IJ.java:181)
	at ij.Executer.runCommand(Executer.java:137)
	at ij.Executer.run(Executer.java:66)
	at java.lang.Thread.run(Thread.java:745)

java.lang.NoSuchMethodException: Could not find method loci.common.DataTools.parseDouble(Ljava/lang/String;)Ljava/lang/Double;
There was a problem with the class loci.common.DataTools which can be found here:
/home/osboxes/Downloads/Fiji.app/jars/bio-formats/formats-common-5.1.10.jar
/home/osboxes/Downloads/Fiji.app/jars/bio-formats/ome-common-5.3.1.jar

WARNING: multiple locations found!
java.lang.NoSuchMethodError: loci.common.DataTools.parseDouble(Ljava/lang/String;)Ljava/lang/Double;
	at loci.formats.in.ZeissCZIReader.translateInformation(ZeissCZIReader.java:2061)
	at loci.formats.in.ZeissCZIReader.translateMetadata(ZeissCZIReader.java:1783)
	at loci.formats.in.ZeissCZIReader.initFile(ZeissCZIReader.java:1028)
	at loci.formats.FormatReader.setId(FormatReader.java:1397)
	at loci.plugins.in.ImportProcess.initializeFile(ImportProcess.java:498)
	at loci.plugins.in.ImportProcess.execute(ImportProcess.java:141)
	at loci.plugins.in.Importer.showDialogs(Importer.java:140)
	at loci.plugins.in.Importer.run(Importer.java:76)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:217)
	at ij.IJ.runPlugIn(IJ.java:181)
	at ij.Executer.runCommand(Executer.java:137)
	at ij.Executer.run(Executer.java:66)
	at java.lang.Thread.run(Thread.java:745)

	at loci.formats.in.ZeissCZIReader.translateInformation(ZeissCZIReader.java:2061)
	at loci.formats.in.ZeissCZIReader.translateMetadata(ZeissCZIReader.java:1783)
	at loci.formats.in.ZeissCZIReader.initFile(ZeissCZIReader.java:1028)
	at loci.formats.FormatReader.setId(FormatReader.java:1397)
	at loci.plugins.in.ImportProcess.initializeFile(ImportProcess.java:498)
	at loci.plugins.in.ImportProcess.execute(ImportProcess.java:141)
	at loci.plugins.in.Importer.showDialogs(Importer.java:140)
	at loci.plugins.in.Importer.run(Importer.java:76)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:217)
	at ij.IJ.runPlugIn(IJ.java:181)
	at ij.Executer.runCommand(Executer.java:137)
	at ij.Executer.run(Executer.java:66)
	at java.lang.Thread.run(Thread.java:745)
```

sbesson, 2017-08-08:
@sebi06: this is a known Fiji specific issue related to a conflict between the `formats-common` JARs shipped in the Fiji update site and the `ome-common` JARs shipped in the Java8 update site - see http://forum.imagej.net/t/fiji-and-bioformats-conflicting-jars/5736.

For now, removing or renaming the old `formats-common` JAR as you did is the best way to select the `ome-common` JAR. As highlighted by Curtis in the thread linked above, the proper resolution of this issue requires upstream Fiji work to selectively activate the update sites depending on the version of Java.

imagejan, 2017-08-09:
@sebi06 wrote:
> I renamed formats-common-5.1.10.jar to formats-common-5.1.10.jar.old

Instead of manually renaming the file, I recommend using the updater to actually uninstall the file:

[image]

That way, it won't suggest to re-install it at the next update 😄

sbesson, 2017-08-10:
@imagejan: just tested the workflow you suggested, but right after the uninstallation of the JAR, the Updater flagged`formats-common` as a candidate for reinstallation (before and after restarting Fiji). Is there a special trick to force an uninstallation step to be remembered?

imagejan, 2017-08-11:
@sbesson you're right, it does so for me as well now. I thought that in the past, it had remembered the choice and not flagged uninstalled files again. And on the other hand, I also remember cases in the past where file were added to an update site but were _not_ automatically flagged for update for some people...

@ctrueden can you comment if this is intended to work, or might have changed recently?
* Flag a file to *Uninstall*
* Apply changes
* Restart Fiji
* Run the updater, the selected file should not be flagged for installation

sbesson, 2017-12-18:
As per http://forum.imagej.net/t/fiji-and-bioformats-conflicting-jars/5736/9, we finally resolved this issues at the update site level with @ctrueden. If the ImageJ, Fiji and Java-8 update sites are activated, there should be no more conflicting jars.

The table below summarizes the expectations in terms of Bio-Formats depending on the combination of enabled update sites:

| ImageJ | Fiji | Fiji-legacy | Java-8 | Bio-Formats                                                      |
|--------|------|-------------|--------|------------------------------------------------------------------|
| yes    | yes  | no          | no     | not functional                                                   |
| yes    | yes  | yes         | no     | 5.1.10 (last Java-6 compatible version)                          |
| yes    | yes  | no          | yes    | 5.x (last release)                                               |
| yes    | yes  | yes         | yes    | 5.x (conflicting jars and potentially other compatiblity issues) |

Closing this issue as we hope it is now resolved but let us know if you encounter more issues.

closed 2017-12-18 by sbesson

ctrueden, 2017-12-18:
Thanks @sbesson!

@imagejan wrote:
> Run the updater, the selected file should not be flagged for installation

Just to respond to this, since I missed it before: I believe that files get flagged for reinstallation if anything they depend on is still installed. So because several Bio-Formats JAR files are marked as depending on `formats-common`, the Updater wants to keep reinstalling it. There is not, to my knowledge, a way to force this to stop happening, except to also uninstall the dependent files.

Fortunately, the issue should be moot now, with the new structure.