ome/bioformats #2924: Uncaught exceptions in PQBinReader, FlowSightReader

mcquin, 2017-08-08:
Hi all,

Here are two uncaught exceptions I see frequently when loading images:

```
Exception in thread "Thread-1" java.io.EOFException: Attempting to read beyond end of file.
	at loci.common.NIOFileHandle.readFloat(NIOFileHandle.java:352)
	at loci.common.RandomAccessInputStream.readFloat(RandomAccessInputStream.java:501)
	at loci.formats.in.PQBinReader.isThisType(PQBinReader.java:103)
Caused by: java.nio.BufferUnderflowException
	at java.nio.Buffer.nextGetIndex(Buffer.java:506)
	at java.nio.HeapByteBuffer.getFloat(HeapByteBuffer.java:463)
	at loci.common.NIOFileHandle.readFloat(NIOFileHandle.java:350)
	... 2 more
Caught exception in class loci.formats.in.PQBinReader.isThisType
```

```
Exception in thread "Thread-1" java.lang.IllegalArgumentException
	at java.nio.Buffer.position(Buffer.java:244)
	at loci.common.NIOFileHandle.buffer(NIOFileHandle.java:605)
	at loci.common.NIOFileHandle.seek(NIOFileHandle.java:293)
	at loci.common.RandomAccessInputStream.seek(RandomAccessInputStream.java:174)
	at loci.formats.tiff.TiffParser.<init>(TiffParser.java:110)
	at loci.formats.in.FlowSightReader.isThisType(FlowSightReader.java:99)
Caught exception in class loci.formats.in.FlowSightReader.isThisType
```

I am able to load images of any supported type. Allowing these exceptions to raise causes a lot of unnecessary noise.

I plan to look into this further in the next few days. However, my Java is quite rusty and I am unfamiliar with the inner workings of this project. If any more experienced contributors have suggestions, I'd love to hear them. :)

Best,
Claire

mcquin, 2017-08-08:
It does say `Caught exception in class`. Specifically, my issue is that these exceptions bypass the logger and always print to `STDOUT`. I am unable to configure a logger which forwards these messages to a log file.

dgault, 2017-08-10:
Hi Claire,

Thanks for raising the issue, looking at those 2 specific readers it does appear that there is the potential for uncaught exceptions within the `isThisType` functions. The `isThisType` will be getting called while trying to determined the correct format and reader to use within Bio-Formats. The exceptions could certainly be caught within the method and simply return a value of false to indicate that the file is not valid for that format.

If you have the time to look at it and open a PR I would be happy to provide guidance, otherwise I can add it to the Bio-Formats inbox to fix for a future release.

David Gault

PR #2929 (2017-08-23): Catch exceptions in isThisType
Resolves #2924
closed 2017-08-23 by sbesson