ome/bioformats #2946: Bioformats issues with Z.1 lightsheet files.

shohpfizer, 2017-08-28:
Hi,

I am currently trying to use bioformats (and python bioformats) to read some CZI files generated by a Z.1 lightsheet. The series is a tiled imaging of a single channel and was done with a dual illumination. The series were separated by illumination, thus, 'file.czi' = left illumination, 'file(1).czi' = right illumination, 'file(2).czi' - left again. (For each stack: XYZTC = 1400,1400,350,1,1)

Now we would like to open these files separately in python/bioformats and extract the image data. For some reason it works well with the files from one side, but gives me an empty (zeros) image for the other illumination. [Zeiss ZEN is displaying both cases correctly.]

Has anyone encountered this issue, and if yes, is there any good fix? I would be happy to just extract the image contents of both illumination images reliably.

Below a screenshow of the file listing, their shapes at a certain z position, and the maximum value of that slice - note that the channel switches from 2 (uneven numbered tile) to 3 (even numbered tile) , and in the 3 channel case, there is no valid image data (max value of zero). Oddly enough, the masterfile (not shown) displays only 2 channels and still has a valid max pixel value.

[image]

Has anyone encountered this issue and knows how to fix that?

Thank you and best regards.

dgault, 2017-08-29:
Hi @shohpfizer,

In order for us to test and debug this issue would it be possible to upload a sample fileset to https://www.openmicroscopy.org/qa2/qa/upload/?

David Gault

shohpfizer, 2017-08-29:
Hi @dgault , since the filelimit was too small, I uploaded it to Box. Please let me know if you got my email.
Best!

dgault, 2017-09-01:
Hi @shohpfizer, I was able to test the provided sample files today and from debugging through the code the issue appears related to handling of the Illuminations dimension metadata. 

For the second set of files the metadata is parsed to show `dimension=I, start=1, size=1, startCoordinate=1.0, storedSize=1`. In this case Bio-Formats has used the start index (0-based) and the size to determined that there is a mismatch and has incorrectly updated the number of Illuminations. This in turn has caused the channel count to increase and the indexing when searching for the planes is also incorrect.

I believe I should be able to put a fix in place to better handle this scenario when the files are being opened individually. Once I have a PR opened I will link it to this issue.

shohpfizer, 2017-09-01:
Thanks @dgault . Yes, it would be great if BF could read any file out of the series independently of the master file. I am looking forward to the development!

dgault, 2017-09-28:
I have opened a PR for review and testing against our existing image repository - https://github.com/openmicroscopy/bioformats/pull/2969