ome/bioformats #2955: ND2 incorrect T, Z sizes

caspervdw, 2017-09-06:
Dear bioformats developers,

I posted this issue on the mailing list yesterday, but today I found this GitHub page. I guess it is useful to have the issue here.

I maintaining a package that wraps a Python interface around Bioformats (part of PIMS, see https://github.com/soft-matter/pims). Some unittests continuously test the correct functioning of the wrapper (and indirectly also of Bioformats). For some time now, Bioformats is failing on a small multidimensional (X, Y, Z, C, T) ND2 file that we use for testing.

There are no issues using Bioformats 5.3.4.

Starting from Bioformats 5.4.0, the following exception is displayed when calling `setId()`:
`java.lang.IndexOutOfBoundsException: Index: 0, Size: 0`

For the newest version (5.7.0) there is no exception at `setId()`, but `getSizeZ()` and `getSizeT()` both return 1 while the should return 10 and 3 respectively.

A link to the file: https://github.com/soft-matter/pims/blob/master/pims/tests/data/bioformats/cluster.nd2
See here for the relevant issue at PIMS: https://github.com/soft-matte
r/pims/pull/266

I hope you can reproduce this issue and that it can be solved in a future version of Bioformats. Thanks for the work on this great project!

sbesson, 2017-09-06:
Hi @caspervdw,

thanks for getting in touch and providing a sample file allowing to reproduce. I can definitely confirm the dimensions and exceptions reported above using the various Bio-Formats command-line tools (versions 5.3.4, 5.4.0 and 5.7.0).

Bio-Formats 5.4.0 introduced a large refactoring of the Nikon ND2 reader to make use of the available chunkmap in order to speed up the reader initialization notably for very large Nikon ND2 files - see https://github.com/openmicroscopy/bioformats/pull/2776.

Although this change has been tested against all Nikon ND2 files in our data repository, we have received reports of [issues](https://trello.com/c/RsDUMigo/7-nd2-chunkmap-issue) caused by this change. Bio-Formats 5.7.0 included a first fix for this behavior -see https://github.com/openmicroscopy/bioformats/pull/2939. From this issue, [cluster.nd2](https://github.com/soft-matter/pims/blob/master/pims/tests/data/bioformats/cluster.nd2) is another sample file which was affected by this change.

As an immediate workaround, disabling the chunkmap processing by setting the [nativend2.chunkmap](http://docs.openmicroscopy.org/bio-formats/5.7.0/formats/options.html) option to `false` should restore the correct dimensionality:

```
$ ./bftools-5.7.0/showinf -nopix cluster.nd2 | grep Size[ZTC]
	SizeZ = 1
	SizeT = 1
	SizeC = 2
PinholeSizeType: 0
$ ./bftools-5.7.0/showinf -nopix -option nativend2.chunkmap false cluster.nd2 | grep Size[ZTC]
	SizeZ = 10
	SizeT = 3
	SizeC = 2
PinholeSizeType: 0
```

The proper fix to this issue involves investigating how the [chunkmap processing code](https://github.com/openmicroscopy/bioformats/blob/v5.7.0/components/formats-gpl/src/loci/formats/in/NativeND2Reader.java#L407) modifies the dimensionality detection. At the moment, we can use this issue to record the problem and track any progress. Having the data publicly available is much appreciated in case anyone in the community want to get involved in fixing this bug.

caspervdw, 2017-09-14:
Thanks @sbesson and good that this issue is looked at.

Could you give me a hint were to find this `nativend2.chunkmap` option in the [API](https://downloads.openmicroscopy.org/bio-formats/5.7.0/api/)? Then I could patch the PIMS reader for the time being. 

Also, I have looked for it before, but is it possible to extract the version from `loci_tools.jar`? Because we need a version conditional for this patch.

caspervdw, 2017-09-14:
I found it myself. In Python:

```
# circumventing the reserved keyword 'in'
mo = getattr(loci.formats, 'in').DynamicMetadataOptions()
mo.set('nativend2.chunkmap', 'False')
reader.setMetadataOptions(mo)
```

And the version: `loci.formats.FormatTools.VERSION`

dgault, 2017-09-14:
Hi @caspervdw, the `nativend2.chunkmap` option is a [format specific option](https://docs.openmicroscopy.org/bio-formats/5.7.0/formats/options.html) which can be set using the [DynamicMetadataOptions](http://downloads.openmicroscopy.org/bio-formats/5.7.0/api/loci/formats/in/DynamicMetadataOptions.html) class. A short of example of how this looks is below:

```
DynamicMetadataOptions options = new DynamicMetadataOptions();
options.setBoolean("nativend2.chunkmap", false);
reader.setMetadataOptions(options);
```

caspervdw, 2017-09-14:
Thanks @dgault!

melissalinkert, 2023-08-11:
Retested with 7.0.0; the default behavior of `showinf` matches what is shown in NIS Elements Viewer (2 channels, 3 T, 10 Z). https://github.com/openmicroscopy/data_repo_config/pull/587 adds the file to the test repository to prevent future regressions.

closed 2023-08-11 by melissalinkert