ome/bioformats #2958: Dll interface for bioformats

ctr26, 2017-09-12:
Hi all,

Is it possible for the cpp wrappers to be compiled as a dll?

sbesson, 2017-09-12:
Hi @ctr26, to be able to answer your question more precisely, can you precise which wrappers you are interested in? There are 2 types of wrappers we can think about:

- the [Bio-Formats JACE bindings](https://github.com/ome/bio-formats-jace) which make the Bio-Formats library accessible from C++. These wrappers have been decoupled from the Bio-Formats code based and are no longer actively maintained by OME since Bio-Formats 5.2.0.
- the former Bio-Formats C++ library which has also been decoupled from the Bio-Formats code base and renamed as [OME Files](https://www.openmicroscopy.org/2016/04/11/ome-files-cpp-0-1-0.html) 

Best,
Sebastien

ctr26, 2017-09-12:
OME Files

ctr26, 2017-09-12:
I'm guessing I need to write essentially a linker between the cpp library within my own dll? Is that correct?

sbesson, 2017-09-13:
Hi,

we currently have no OME Files DLLs and would need to address a couple of technical blockers although this is something we want to look into in the future.

At the moment, we are producing various binary Windows build [for each release of OME Files](http://www.openmicroscopy.org/ome-files/downloads/) which can be used by downstream consumers. See e.g. https://github.com/openmicroscopy/ome-files-performance#windows for an example of consumption of these bundles by our benchmarking component.

Hope this helps a bit in your case,
Sebastien

ctr26, 2017-09-21:
Great news, if OME were implemented as a DLL I would be happy to write a LabVIEW wrapper

joshmoore, 2017-09-22:
@ctr26 : if you mean a LabVIEW wrapper to read and write OME-TIFFs, it should be possible to begin on that now.

ctr26, 2017-09-22:
How so? From what I've read LabVIEW has limited ability to interface with other languages.

joshmoore, 2017-09-22:
I may be misunderstanding, but you could compile a shared C++ library today to read/write OME-TIFF.

ctr26, 2017-09-22:
Oh really? Would I just have to write a shared library which would call the cpp functions from ome-tiff?

joshmoore, 2017-09-22:
Well, I would think you link against the ome-files DLL. Does something like https://github.com/ome/ome-qtwidgets/blob/master/docker/c7/Dockerfile help you get started?

sbesson, 2017-11-28:
Closing this as this is unrelated to Bio-Formats. Feel free to open a new issue against https://github.com/ome/ome-files-cpp.

closed 2017-11-28 by sbesson