ome/bioformats #2962: Stitch tiles on a .czi file

yere1, 2017-09-19:
Hello,
My .czi file is made up of 2D tiles, each of the tiles being a Z-stack. ImageJ has no problem stitching those tile together to make a single larger image. However, by using bfopen(), I could not find any metadata that would help me do the same in MATLAB: all bfopen does is to return each tile as a separate plane. I have attached the file so that you can take a look.

Thanks!
GoogleDrive link: https://drive.google.com/file/d/0BzkDz7cWMKIuWmNVdXpXYjNxbFU/view?usp=sharing

sbesson, 2017-09-20:
Hi @yere1, thanks for getting in touch and sharing a file to reproduce the issue. Trying to open the CZI file under Fiji directly via the Bio-Formats importer, I can confirm the image is loaded as individual multi-dimensional 512x512 tiles even with the CZI autostitch option turned on. This layout is consistent with what is returned by `bfopen` in MATLAB.

How do you perform the stitching of those tiles into a larger image. Are you using a specific plugin like the [Stitching plugin](http://imagej.net/Image_Stitching)?

yere1, 2017-09-20:
Thank you for the response. Here is a recording of the screen as I open the file, so that you can see how I open it on Fiji. The version for ImageJ I use is 2.0.0 from 23/02/2017. 

Link  to Video: https://drive.google.com/open?id=0BzkDz7cWMKIuWmJSZGgteFZCcms

sbesson, 2017-09-21:
@yere1 thanks for the detailed workflow video. The main difference I can see is the `Stitch tiles` option enabled in the Bio-Formats importer. In terms of API, this checkbox effectively adds the [TileStitcher](http://downloads.openmicroscopy.org/bio-formats/5.7.1/api/loci/formats/TileStitcher.html) wrapper to the stack while opening your CZI file.

As this reader wrapper is not available in`bfopen`, to open the image files in the same way, you will need to initialize a reader using the low-level methods as follows:

```
bfCheckJavaPath();
bfInitLogging;
r = loci.formats.TileStitcher(bfGetReader());
r.setId(path_to_your_file);
```

and then use the rest of `bfopen.m` as usual to access individual planes.

yere1, 2017-09-21:
Thank you for help! Just one more question: how exactly do I have to implement those 4 lines of code? Where exactly within **bfopen()** do I have to add those 4 lines? I tried copy-pasting them after line 114, where it says **r = bfGetReader(id, stitchFiles);**, and got error messages.

sbesson, 2017-09-22:
If you are modifying `bfopen.m` or a copy of it, can you try replacing

```
 r = bfGetReader(id, stitchFiles);
```

by

```
r = loci.formats.TileStitcher(bfGetReader());
r.setId(id);
```

yere1, 2017-09-22:
I tried, and got the following error message: 

>>> bfopen
Z positions not equal
Error using bfopen (line 115)
Java exception occurred:
java.lang.NullPointerException
	at ome.units.quantity.Length.compareTo(Length.java:150)
	at ome.units.quantity.Length.compareTo(Length.java:35)
	at java.util.ComparableTimSort.countRunAndMakeAscending(ComparableTimSort.java:290)
	at java.util.ComparableTimSort.sort(ComparableTimSort.java:157)
	at java.util.ComparableTimSort.sort(ComparableTimSort.java:146)
	at java.util.Arrays.sort(Arrays.java:472)
	at loci.formats.TileStitcher.setId(TileStitcher.java:290)

sbesson, 2017-09-25:
Hi @yere1, could you tell us which version of Bio-Formats is used in Fiji (`Help > About Plugins... > Bio-Formats plugin`) and in MATLAB (`[~,version]=bfCheckJavaPath()`)?

yere1, 2017-09-25:
For MATLAB, it is 5.5.3; for Fiji, it is 5.5.0

sbesson, 2017-09-25:
Thanks for the versions information. By looking at the output of `showinf -nopix -omexml` using your file using both versions, Bio-Formats 5.5.0 reads the `PositionX` and `PositionY` of all images in the `reference frame` unit while Bio-Formats 5.5.3 reads the `PositionX` and `PositionY` in `micron` for the first image and `reference frame` for the following images. This unit mismatch is in agreement with the error in https://github.com/openmicroscopy/bioformats/issues/2962#issuecomment-331558689.

It seems this behavior change might be related to https://github.com/openmicroscopy/bioformats/pull/2881 which attempted to fix the way CZI positions are parsed in Bio-Formts 5.5.3. We will compare with the expectations reported from ZEN to confirm whether these changes might have introduced a regression for some CZI modalities. /cc @melissalinkert 

In the meantime, you should be able to workaround the issue by using the same version of Bio-Formats, 5.5.2 or lower, in Fiji and MATLAB.

melissalinkert, 2017-09-25:
Screenshots from ZEN:

[image]
[image]

Z positions reported by ZEN (in micrometers) are: 0, 2.92, 5.85, 8.77, 11.7, 14.62, 17.54, 20.47, 23.39, 26.32, 29.24, 32.16, 35.09, 38.01, 40.94, 43.86

That's definitely not what I see with ```showinf -nopix -omexml``` using 5.7.1.  Only the first series has ```PositionZ```, and for every plane it is ```PositionZ="-107.917" PositionZUnit="µm"```.  The ```PositionX``` and ```PositionY``` are correct for the first series, but as @sbesson notes the units change for subsequent series. I have a private branch that may fix the positions; will try it a little later today and open a PR if appropriate.

Otherwise, ZEN is reporting a single position, not 16 positions, so I suspect the underlying issue here may be related to https://trello.com/c/7MRDBwSn/178-czi-file-not-automatically-stitching (i.e. it may be a bug that ```TileStitcher``` is required at all in this case).

yere1, 2017-09-25:
Thank you for the detailed explanation. I tried it with BF 5.5.2 and it worked!
Thanks a lot for your help,
Yere1

sebi06, 2017-09-25:
Hi guys, since iam responsive at Zeiss for the CZI I am a bit confused (maybe I did get something wrong).
What is the issues with the zposition inside the czi that are dissent from the ones when reading this czi with bioformats?
And may I have a look at that CZI?
Sebi