ome/bioformats #2971: Contribution of File Format documentation / code for Bio-Rad Gel format (1sc)

itsayellow, 2017-10-09:
I tried to post to the Bio-Formats forum, but in case the developers have not seen it, here is some info (relevant to https://docs.openmicroscopy.org/bio-formats/5.6.0/formats/bio-rad-gel.html):

Bio-Rad Gel Format specification:
https://github.com/itsayellow/biorad1sc_doc/blob/master/file_1sc_spec.rst

Python implementation of a Bio-Rad Gel Format reader:
https://github.com/itsayellow/biorad1sc_reader
https://pypi.org/project/biorad1sc-reader/
http://biorad1sc-reader.readthedocs.io/en/latest/
`pip3 install biorad1sc_reader`

sbesson, 2017-10-09:
Cross-linking to the relevant [forum post](https://www.openmicroscopy.org/community/viewtopic.php?f=13&t=8365&p=18508&hilit=itsayellow#p18508). Community-driven specifications/implementations are definitely great to see in this ecosystem.

As this is open as an issue, what would be the next steps here @itsayellow. One possibility would be to link to these resources in the format documentation page? All format pages are generated from [format-pages.txt](https://github.com/openmicroscopy/bioformats/blob/develop/components/autogen/src/format-pages.txt) (see [the doc for more information](http://docs.openmicroscopy.org/bio-formats/5.7.1/developers/format-documentation.html)).

Would you consider opening a Pull Request registering the spec/implementation under `software`?

itsayellow, 2017-10-09:
Sure thing, I can do a pull-request for the documentation.  Mostly I was assuming this would help as a resource (documentation.)  I understand as far as code goes you are using Java and my code is Python.

I'll start figuring out how to fork/pull-request on the documentation entry.

sbesson, 2017-10-10:
Thanks for opening the Pull Request. While the code is definitely Java only, there are a couple of community maintained Python projects wrapping Bio-Formats (https://github.com/CellProfiler/python-bioformats/, https://github.com/soft-matter/pims). The format documentation pages are a good place to link to valuable third-party resources with no particular language restriction.

sbesson, 2017-11-28:
Closing as this contribution is now referenced from the [Bio-Rad documentation page](https://docs.openmicroscopy.org/bio-formats/5.7.2/formats/bio-rad-gel.html). Thanks @itsayellow.

closed 2017-11-28 by sbesson