ome/bioformats #2982: Avizo binary files imported as text (Fiji/ImageJ)

iarganda, 2017-10-26:
Avizo (AmiraMesh) binary files are imported in Fiji/ImageJ as text files. It seems to be a simple problem associated with the new header of the software because manually changing the first line of the file header from something like

`# Avizo BINARY-LITTLE-ENDIAN 2.1`

to

`# AmiraMesh BINARY-LITTLE-ENDIAN 2.1`

makes it work.

rleigh-codelibre, 2017-10-26:
@iarganda Thanks for reporting this.  Would it be possible to provide a sample file using this new header at http://qa.openmicroscopy.org.uk/qa/upload/ ?  It looks like a simple change to support both the old and new variants, but having a file to validate it works will ensure the fix works and won't cause any regressions.

Thanks,
Roger

iarganda, 2017-10-26:
@rleigh-codelibre I could not use your site but [here](https://ehubox.ehu.eus/s/p8atohaip8b69Jy) you are a very small example.

rleigh-codelibre, 2017-10-26:
@iarganda Thanks.  This contains `Avizo 3D BINARY 2.0` rather than the header you mentioned above.  Is this correct or is it a multi-file set or a different file type?

iarganda, 2017-10-26:
@rleigh-codelibre It is correct, that header also provokes the import as text.

rleigh-codelibre, 2017-10-26:
@iarganda  Do you have a sample file using the `Avizo BINARY-LITTLE-ENDIAN 2.1` in addition though, so I can validate the parsing of that header?

iarganda, 2017-10-26:
@rleigh-codelibre  Not a one I can share but you can use the same file and manually edit the header. It should be the same.

rleigh-codelibre, 2017-10-26:
Please see the linked PR for a suggested fix.  If you would like to try this out, it would be useful to know if this fixes the problem for you.

sbesson, 2017-10-26:
@iarganda: thanks again for the bug report and the sample file. To help the community getting involved with the maintenance of Bio-Formats, we are increasingly trying to make our sample data publicly accessible. Can the file uploaded in https://github.com/openmicroscopy/bioformats/issues/2982#issuecomment-339664345 be made public at https://downloads.openmicroscopy.org/images/ under a CC-BY license? Otherwise, we will keep it private but add them to our curated data repository against which each Bio-Formats change is tested.

iarganda, 2017-10-27:
@sbesson sure! It is just the public bat cochlea volume saved as AmiraMesh.

iarganda, 2017-10-27:
@rleigh-codelibre I confirm the PR fixes the problem. Thanks!

PR #2984 (2017-12-19): AmiraReader: Handle 'Avizo' in the file header
Closes #2982

Allow `AmiraMesh` or `Avizo` in the file header.

Testing: see #2982 for sample file.

Test configuration in https://github.com/openmicroscopy/data_repo_config/pull/245
closed 2017-12-19 by sbesson