ome/bioformats #3007: GlobalMetadata of .fake test files contains "header" field

aoles, 2017-11-23:
Hi,
When I run run `./showinf` on **testfile.fake.ini**
```
[GlobalMetadata]
key=value
```
I get the following.
```
Initializing reader
FakeReader initializing testfile.fake
Initialization took 0.03s

Reading core metadata
filename = testfile.fake
Used files:
        /home/oles/Downloads/bftools/testfile.fake
        /home/oles/Downloads/bftools/testfile.fake.ini
Series count = 1
Series #0 :
        Image count = 1
        RGB = false (1) 
        Interleaved = false
        Indexed = false (true color)
        Width = 512
        Height = 512
        SizeZ = 1
        SizeT = 1
        SizeC = 1
        Thumbnail size = 128 x 128
        Endianness = intel (little)
        Dimension order = XYZCT (certain)
        Pixel type = uint8
        Valid bits per pixel = 8
        Metadata complete = true
        Thumbnail series = false
        -----
        Plane #0 <=> Z 0, C 0, T 0

Reading global metadata
header: GlobalMetadata
key: value

Reading metadata
```

Note the additional field `header: GlobalMetadata`. Is this intended?

dgault, 2017-11-23:
Yes this is to be expected. The particular behaviour of the `IniParser`  is to check for 2 types of header, a chapter header with `{}` and a section header with `[]`. The section header will be stored as a key value pair with `header` as the key.

aoles, 2017-11-23:
Thanks a lot for the clarification! I found the recommendation to introduce global metadata by using `[GlobalMetadata]` in [Bio-Formats documentation](https://docs.openmicroscopy.org/bio-formats/5.7.2/developers/generating-test-images.html).

I was wondering about the following as well. First of all, is there a way to specify series metadata in test images? And secondly, is is possible to provide separate core metadata for each image series such that they are not identical but have, for example, different dimensions?

Cheers,
Andrzej

dgault, 2017-11-24:
Hi Andrzej,

If we are talking specifically about the .fake test files then I am afraid the core metadata will be duplicated for each series rather than being able to specify unique values. However if you need help series specific core metadata for other formats please do let us know and I can assist further.

Also for the fake format any key value pairs undefined in https://docs.openmicroscopy.org/bio-formats/5.7.1/developers/generating-test-images.html will be added to the global metadata with no differentiation for series metadata.

aoles, 2017-11-24:
Thanks for your reply David. Re the global metadata in .fake test files: is it then possible to somehow avoid the creation of the spurious header field? I tried specifying a chapter header `{GlobalMetadata}` instead of a section header but this doesn't seem to work at all:

```
> echo -e "{GlobalMetadata}\nkey=value" > testfile.fake.ini; showinf -nopix testfile.fake.ini
Checking file format [Simulated data]
Initializing reader
FakeReader initializing testfile.fake.ini
Initialization took 0.025s

Reading core metadata
filename = testfile.fake
Used files:
        /home/oles/Downloads/bftools/testfile.fake
        /home/oles/Downloads/bftools/testfile.fake.ini
Series count = 1
Series #0 :
        Image count = 1
        RGB = false (1) 
        Interleaved = false
        Indexed = false (true color)
        Width = 512
        Height = 512
        SizeZ = 1
        SizeT = 1
        SizeC = 1
        Thumbnail size = 128 x 128
        Endianness = intel (little)
        Dimension order = XYZCT (certain)
        Pixel type = uint8
        Valid bits per pixel = 8
        Metadata complete = true
        Thumbnail series = false
        -----
        Plane #0 <=> Z 0, C 0, T 0

Reading global metadata

Reading metadata
```

dgault, 2017-11-28:
Hi Andrzej,

Firstly regarding the spurious header field, other readers do ignore this field (such as in https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-gpl/src/loci/formats/in/LiFlimReader.java#L305) however the FakeReader currently does not. This is something which could be considered a minor bug and potentially changed in a future patch release. If this is a fix which you require urgently and wished to open a PR to resolve I would be happy to assist you in doing so and review the changes.

For the different annotation types, generally speaking series metadata will be a generic way to store and preserve format specific metadata values which do not fit into the OME Model. The other various type specific annotations are there to provide a more customised usage for scenarios in which specific metadata will likely be written and then read back and reused. 

In order to read various different annotation types you would use `MetadataRetrieve` similar to the below sample:

```
MetadataRetrieve retrieve = service.asRetrieve(reader.getMetadataStore());
for (int i=0; i<retrieve.getBooleanAnnotationCount(); i++) {
      String id = retrieve.getBooleanAnnotationID(i);
      String id = retrieve.getBooleanAnnotationNamespace(i);
      Boolean value = retrieve.getBooleanAnnotationValue(i);
}
```

Which type of annotation is best will likely depend on the type of data you are storing and how it is being used. Do you have an idea of what sort of metadata are you expecting to store in the Annotations and how you are hoping to use it once its been read back?

sbesson, 2018-03-14:
This issue has been addressed and has no further activity. It is therefore being closed. Thank you
for your contributions, they are much appreciated.

closed 2018-03-14 by sbesson