ome/bioformats #3008: Setters for global metadata and series metadata

aoles, 2017-11-23:
Hi,
*IFormatReader* contains getters for retrieving bulk lists of global metadata and series metadata: `getGlobalMetadata` and `getSeriesMetadata`, respectively. Is there a way of going in the opposite direction? Supposed I have a hashtable containing the relevant metadata field/value pairs, how would I pass it to the *MetadataRetrieval* object associated with an *IFormatWriter*? So far I've only managed to add core metadata through a call to `MetadataTools.populateMetadata`. 

Any suggestions are highly appreciated.

Cheers,
Andrzej

dgault, 2017-11-23:
Hi Andrzej,

Collectively we refer to both series and global metadata as original metadata. Both of these types are stored in the OME-XML as XMLAnnotations, specifically as an OriginalMetadataAnnotation.

An example of how these annotations are written is as below:

```
ServiceFactory factory = new ServiceFactory();
OMEXMLService service = factory.getInstance(OMEXMLService.class);
OMEXMLMetadata metadata = service.createOMEXMLMetadata();
IFormatWriter writer = new ImageWriter();
Hashtable<String, Object> originalMetadata = null;
//set metadata
service.populateOriginalMetadata(metadata, originalMetadata);
writer.setMetadataRetrieve(metadata);
```

If you are writing both series and global metadata then the way Bio-Formats writes these values is that both sets of metadata are prefixed with the series name and merged together before being written. An example of how this might look is as below:

```
Hashtable<String, Object> globalMetadata = new Hashtable<String, Object>();
allMetadata.putAll(globalMetadata);

for (int series=0; series<getSeriesCount(); series++) {
  String name = "Series " + series;
  String realName = ((IMetadata) store).getImageName(series);
  if (realName != null && realName.trim().length() != 0) {
    name = realName;
  }
  setSeries(series);
  MetadataTools.merge(getSeriesMetadata(), allMetadata, name + " ");
  setSeries(0);
}
setSeries(0);
service.populateOriginalMetadata((OMEXMLMetadata) store, allMetadata);
```

David

aoles, 2017-11-23:
Thanks a lot David for your comprehensive answer an for providing these code samples, this is very helpful.

aoles, 2017-11-25:
Hi David,

I tried setting global metada stored in a `Hashtable<String, Object>` by passing it to `service.populateOriginalMetadata`, however, after saving and reloading the file metadata types are not preserved and all end up as strings. Is this expected or am I missing something?

Cheers,
Andrzej

dgault, 2017-11-28:
Hi Andrzej,

Yes is the expected behaviour. The `OriginalMetadataAnnotation` will store both the key and value as String. When called via `populateOriginalMetadata` this will simply use the `toString()` method when saving when the values.

aoles, 2017-11-30:
Thanks David.

Could you then maybe advise on an alternative way of adding metadata such that their types are preserved rather than converted to `String`?

Cheers,
Andrzej

dgault, 2017-12-05:
Hi Andrzej,

To preserve the type you can use the other type specific annotations, though you will have to be aware to retrieve them directly rather than through the `populateOriginalMetadata` and `getOriginalMetadata` functions.

To write for example a value of type double you would set a `DoubleAnnotation` using the `MetadataStore` interface:
```
      annotationID = "Annotation:" + annotationCount;
      store.setDoubleAnnotationID(annotationID, annotationDoubleCount);
      store.setDoubleAnnotationNamespace("annotationNamespace", annotationDoubleCount);
      store.setDoubleAnnotationValue(value, annotationDoubleCount);
      store.setImageAnnotationRef(annotationID, imageIndex, annotationRefCount);
```

To retrieve it you can then use the `MetadataRetrieve` interface:
```
      int doubleAnnotationID = retrieve.getDoubleAnnotationCount();
      for (int i = 0; i < doubleAnnotationID; i ++) {
         String annotationID = retrieve.getDoubleAnnotationID(i);
         String annotationNamespace = retrieve.getDoubleAnnotationNamespace(i);
         Double annotationValue = retrieve.getDoubleAnnotationValue(i);
      }
```

sbesson, 2018-03-14:
This issue has been addressed and has no further activity. It is therefore being closed. Thank you
for your contributions, they are much appreciated.

closed 2018-03-14 by sbesson