ome/bioformats #3016: No class loci.formats.services.OMEXMLServiceImpl can be located on the Java class path

jkfindeisen, 2017-12-07:
I use Matlab2017b on Windows 10 and downloaded the Bio-Formats Matlab toolbox 5.7.2 from the webpage (https://downloads.openmicroscopy.org/bio-formats/5.7.2/artifacts/bfmatlab.zip). I unzipped it, added it to the path and executed two lines from the documentation.

```
plane = zeros(64, 64, 1, 2, 2, 'uint8');
metadata = createMinimalOMEXMLMetadata(plane);
```

I get an error

```
Error using javaObject
No class loci.formats.services.OMEXMLServiceImpl can be located on the Java class path

Error in createMinimalOMEXMLMetadata (line 51)
OMEXMLService = javaObject('loci.formats.services.OMEXMLServiceImpl');
```
`bfCheckJavaPath` returns true. Neither in the online documentation nor here in the issues I can find anything related.

sbesson, 2017-12-07:
Hi @jkfindeisen,

thanks for reporting this issue. I could not reproduce on MATLAB 2017b, Mac OSX 10.11 and Bio-Formats MATLAB 5.7.2:

```
>> ver('MATLAB')
-----------------------------------------------------------------------------------------------------
MATLAB Version: 9.3.0.713579 (R2017b)
MATLAB License Number: 874227
Operating System: Mac OS X  Version: 10.12.6 Build: 16G1036 
Java Version: Java 1.8.0_121-b13 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.3         (R2017b)
>> [~,v]=bfCheckJavaPath

v =

    '5.7.2'

>> plane = zeros(64, 64, 1, 2, 2, 'uint8');
metadata = createMinimalOMEXMLMetadata(plane)

metadata =

loci.formats.ome.OMEXMLMetadataImpl@41eb94bc
```

To help us narrow down the source of the issue, you tell us more about:
- the Bio-Formats version reorted by `[~, version]=bfCheckJavaPath()`
- the various toolboxes installed and their toolbox installed as reported by `ver`
- the content of your dynamic Java classpath reported by ` javaclasspath('-dynamic')`

jkfindeisen, 2017-12-08:
I played around a bit. The dynamic Java classpath is empty before `bfCheckJavaPath` is called. Afterwards it's filled with the bioformats_package.jar. If `createMinimalOMEXMLMetadata` is the first function that is called from bfmatlab, it will produce this error. You could try for yourself by leaving out the `bfCheckJavaPath` in your example.

I can call `bfCheckJavaPath` once myself to fix this for me and probably the same lines as in the beginning of `bfsave` could also be in `createMinimalOMEXXMLMetadata`.

sbesson, 2017-12-08:
@jkfindeisen thanks for the additional information. At the moment only the high-level utility functions like `bfopen`, `bfsave` are  calling `bfCheckJavaPath` internally.

When using either the low-level MATLAB functions like `bfGetReader`, `createMinimalOMEXXMLMetadata` or the Java API directly, the expectation is that the consumer code will invoke `bfCheckJavaPath` or put the Bio-Formats JARs  in the classpath (static or dynamic) by another mean.

Reading https://docs.openmicroscopy.org/bio-formats/5.7.2/developers/matlab-dev.html in the light of this issue, there is certainly no mention of this expectation. Adding `bfCheckJavaPath` to the low-level methods is a possibility but we would need to review the overhead. Immediately, we can certainly expand the MATLAB documentation to mention the usage of `bfCheckJavaPath`.

jkfindeisen, 2017-12-08:
That sounds reasonable to me. I guess the issue here can be closed.

closed 2017-12-11 by sbesson