ome/bioformats #3018: bfMatlab's bfsafe might not respect dimension order

jkfindeisen, 2017-12-08:
Matlab2017b on Windows 10 and Bio-Formats [Matlab toolbox 5.7.2](https://downloads.openmicroscopy.org/bio-formats/5.7.2/artifacts/bfmatlab.zip).

I use 

```
data = ones(10,20,30);
bfsave(data, 'test.ome.tiff', 'dimensionOrder', 'XYZCT');
```

then I load it with an updated Fiji (*File>Import>Bio-Formats*) and the import dialog of Bio-Formats displays the stack order as 'XYCZT' and doesn't let me change it (greyed out). The displayed OME metadata XML tree shows DimensionOrder='XYCZT'. The image properties seem to be okay though and the viewer shows z slices.

What happens here? Have the dimensions been reordered by `bfsave` or is ImageJ only capable of displaying 'XYCZT' order?

sbesson, 2017-12-11:
@jkfindeisen: yes the `XYCZT` dimension order is a the default one used by the ImageJ Hyperstack but any OME-TIFF file should be handled transparently independently on the underlying `DimensionOrder`

You can verify the dimensions passed to `bfsave` have been digested correctly by reopening the saved file  in MATLAB and introspecting the metadata:

```
>> bfsave(data, 'test.ome.tiff', 'dimensionOrder', 'XYZCT');
>> r=bfGetReader('test.ome.tiff');
>> r.getDimensionOrder()

ans =

XYZCT

>> bfsave(data, 'test.ome.tiff', 'dimensionOrder', 'XYCZT');
>> r=bfGetReader('test.ome.tiff');
>> r.getDimensionOrder()

ans =

XYCZT
```

jkfindeisen, 2017-12-11:
That fully clarifies what is going on there. I was a little bit confused by ImageJ reordering my data, but now I'm fine with it. Thanks.

closed 2017-12-11 by jkfindeisen