ome/bioformats #3023: Opening .czi file creates java exception

demolitionb, 2017-12-15:
Opening some of my .czi files creates the following error:

(Fiji Is Just) ImageJ 2.0.0-rc-61/1.51s; Java 1.8.0_66 [64-bit]; Windows 10 10.0; 93MB of 12146MB (<1%)
 
java.lang.IndexOutOfBoundsException: Index: 1, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:653)
	at java.util.ArrayList.get(ArrayList.java:429)
	at ome.xml.model.OME.getROI(OME.java:738)
	at ome.xml.meta.OMEXMLMetadataImpl.setLineID(OMEXMLMetadataImpl.java:10241)
	at ome.xml.meta.FilterMetadata.setLineID(FilterMetadata.java:1789)
	at loci.formats.in.ZeissCZIReader.populateLines(ZeissCZIReader.java:2855)
	at loci.formats.in.ZeissCZIReader.translateLayers(ZeissCZIReader.java:2536)
	at loci.formats.in.ZeissCZIReader.translateMetadata(ZeissCZIReader.java:1874)
	at loci.formats.in.ZeissCZIReader.initFile(ZeissCZIReader.java:1110)
	at loci.formats.FormatReader.setId(FormatReader.java:1397)
	at loci.plugins.in.ImportProcess.initializeFile(ImportProcess.java:498)
	at loci.plugins.in.ImportProcess.execute(ImportProcess.java:141)
	at loci.plugins.in.Importer.showDialogs(Importer.java:140)
	at loci.plugins.in.Importer.run(Importer.java:76)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:221)
	at ij.IJ.runPlugIn(IJ.java:185)
	at ij.IJ.runPlugIn(IJ.java:174)
	at HandleExtraFileTypes.openImage(HandleExtraFileTypes.java:499)
	at HandleExtraFileTypes.run(HandleExtraFileTypes.java:72)
	at ij.IJ.runUserPlugIn(IJ.java:221)
	at ij.IJ.runPlugIn(IJ.java:185)
	at ij.IJ.runPlugIn(IJ.java:174)
	at ij.io.Opener.openWithHandleExtraFileTypes(Opener.java:515)
	at ij.io.Opener.openImage(Opener.java:369)
	at ij.io.Opener.openImage(Opener.java:242)
	at ij.io.Opener.open(Opener.java:109)
	at ij.io.Opener.openAndAddToRecent(Opener.java:291)
	at ij.plugin.DragAndDrop.openFile(DragAndDrop.java:192)
	at ij.plugin.DragAndDrop.run(DragAndDrop.java:159)
	at java.lang.Thread.run(Thread.java:745)

Some other files are working from the same set of measurements. One not working file is attached. Machine is a Zeiss Orion NanoFab. 

[example-file.zip](https://github.com/openmicroscopy/bioformats/files/1562722/hzdr-sims.zip)

demolitionb, 2017-12-18:
Fix from #3025 didn't change anything regarding this. Still not working.

sbesson, 2017-12-18:
@demolitionb thanks for the early feedback. @dgault opened #3025 right before going on leave hoping this would be a quick fix but we have not performed any proper code or functional review yet.

What was the building and testing workflow you used led to get to https://github.com/openmicroscopy/bioformats/issues/3023#issuecomment-352360169?

I minimally reproduced the initial issue using the Bio-Formats 5.7.2 command line tools:

```
[sbesson@ome-c6100-1 ~]$ ./bftools-5.7.2/showinf -version
Version: 5.7.2
Build date: 21 November 2017
VCS revision: 6909f51c1786b58fe5a0765cea2ad6a278ebc1c2
[sbesson@ome-c6100-1 ~]$ ./bftools-5.7.2/showinf -nopix -omexml hzdr-sims.czi
Checking file format [Zeiss CZI]
Initializing reader
ZeissCZIReader initializing hzdr-sims.czi
Exception in thread "main" java.lang.IndexOutOfBoundsException: Index: 1, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:635)
	at java.util.ArrayList.get(ArrayList.java:411)
	at ome.xml.model.OME.getROI(OME.java:738)
	at ome.xml.meta.OMEXMLMetadataImpl.setLineID(OMEXMLMetadataImpl.java:10241)
	at ome.xml.meta.FilterMetadata.setLineID(FilterMetadata.java:1789)
	at loci.formats.in.ZeissCZIReader.populateLines(ZeissCZIReader.java:2855)
	at loci.formats.in.ZeissCZIReader.translateLayers(ZeissCZIReader.java:2536)
	at loci.formats.in.ZeissCZIReader.translateMetadata(ZeissCZIReader.java:1874)
	at loci.formats.in.ZeissCZIReader.initFile(ZeissCZIReader.java:1110)
	at loci.formats.FormatReader.setId(FormatReader.java:1397)
	at loci.formats.ImageReader.setId(ImageReader.java:839)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1031)
	at loci.formats.tools.ImageInfo.main(ImageInfo.java:1117)
```

Using our internal merge build build which includes the change from #3025, I could not reproduce the problem any more:

```
[sbesson@ome-c6100-1 ~]$ ./bftools/showinf -version
Version: 5.7.3-SNAPSHOT
Build date: 18 December 2017
VCS revision: d0eb5adb25a996d5c854e066df232efe81c6be20
[sbesson@ome-c6100-1 ~]$ ./bftools/showinf -nopix -omexml hzdr-sims.czi
Checking file format [Zeiss CZI]
Initializing reader
ZeissCZIReader initializing hzdr-sims.czi
Parsing schema path
http://www.openmicroscopy.org/Schemas/OME/2016-06/ome.xsd
Validating OME-XML
No validation errors found.
Initialization took 1.909s

Reading core metadata
filename = hzdr-sims.czi
Series count = 1
Series #0 -- hzdr-sims.czi #1:
	Image count = 1
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 512
	Height = 512
	SizeZ = 1
	SizeT = 1
	SizeC = 1
	Thumbnail size = 128 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Reading global metadata
Appliance|Data|Device|Id #1: ETDetector
...
```

demolitionb, 2017-12-18:
Hi @sbesson , 

so I used the 64-bit FIJI package with ImageJ 1.51s, Java 1.8.0_66, BioFormats 5.7.2 ... if that helps.
To test #3025 , I git cloned this 5.7.3 with the proper commit number and checked the src code in ZeissCZIReader.java if there include the changes from #3025.

After that, I compiled everything in Linux Mint with 'ant release' and it confirmed with "build successful" ,... and copied the bioformats-xxx.jar from the home/.m2/xxx folder into my FIJI plugins folder.

demolitionb, 2017-12-18:
Ah, just found that fiji uses bioformats from \Fiji.app\jars and not just from plugins... so probably it still uses the old libraries....

sbesson, 2017-12-18:
@demolitionb: I was about to say so. There are 2 types of JARs: `bioformats_plugins` under `plugins` and the other JARs under `jars/bio-formats`. In your case, if you are modifying 5.7.2 with the changes from #3025 only, the only JAR you want to replace in your distribution is `formats-gpl`.

demolitionb, 2017-12-18:
Just found it too. Thank you and thanks to @dgault ... now it works like a charm! 

So, should I close here? I think it should be included in 5.7.3. ... Thanks again for the fast response.

sbesson, 2017-12-18:
Great. Glad to hear this news and really all the credit goes to @dgault for figuring out the fix.

No need to close this issue, we will link it to the corresponding PR and it will be auto-closed when the latter is merged.

With regard to timelines, assuming everything passes review and does not create any unwanted side-effects on other CZI datasets, #3025 would probably be a candidate for inclusion in Bio-Formats 5.7.3. We will review our timelines in the new year.

sebi06, 2017-12-19:
This is a CZI from ZEN Yellow (a ZEN Blue derivate) running the Helium-Ion-Microscopes. Should of course run "as good" as any other CZI, which is read by the CZIReader from BioFormats. Let me know, if you need help from within ZEISS when it comes to question regarding the data structure.

sbesson, 2017-12-19:
Thanks @sebi06, we will certainly ping you if we need additional information on the data structure or a ZEN color while reviewing #3025.

PR #3025 (2018-01-11): CZI: Allow layers that do not create ROI
Fixes #3023

Exception was being thrown when a parsed layer did not create an ROI. The next time an ROI is added the index used for the ROI was then incorrect. Further investigation may be needed to determined if the skipped layer should indeed be creating an ROI of some form.

To test:
- Ensure builds and tests remain green
- Use the sample file from the associated issue
- Without the PR, read the file from https://github.com/openmicroscopy/bioformats/issues/3023 and verify that an IndexOutOfBoundsException is thrown
- With the PR, read the file from h [... cut]
closed 2018-01-11 by dgault