ome/bioformats #3034: Memoizer should use drive letter to determine folder on Windows

mkitti, 2018-01-09:
Memoizer strips off the drive letter on Windows, but this creates a potential situation where the the memo file may be mistakenly used for the a similarly named file under a similar path but on a different drive.

Suppose the directory property of Memoizer is set to C:\temp
File A is located at C:\somedir\foo.nd2
File B is located at D:\somedir\foo.nd2

Memoizer will use the same memo file for A and B located at C:\temp\somedir\.foo.nd2.bfmemo
This is because the drive is simply stripped:
https://github.com/openmicroscopy/bioformats/blob/master/components/formats-bsd/src/loci/formats/Memoizer.java#L794

I suggest that the memo files should be created at
C:\temp\somedir\C\.foo.nd2.bfmemo
C:\temp\somedir\D\.foo.nd2.bfmemo

That can be accomplished by just removing ":" rather than stripping off the drive letter.

rleigh-codelibre, 2018-01-10:
Rather than mangling the path and have the risk of name clashes, have we considered hashing the path with e.g. SHA2 and using that as the memo filename?  It does have the disadvantage of not being easy to clean up old memo files since the name is obscured, but we generally want to wipe out the memo files wholesale on upgrade, so it might not be an issue.

mkitti, 2018-01-10:
Hashing sound reasonable to me. It may also be useful to include the version number in case a user is switching between versions of bioformats. If you store the pre-hashed paths in some file then it would not be difficult to map the the paths to the hashed.

Our currently scenario is that the image files are located on a write-protected shared network filesystem. This means:
1. The bfmemo file cannot be replaced in the same directory as the the image file itself.
2. Users may be accessing the same image files using different versions of bioformats.
3. bfmemo files thus need to be stored on a user-specific storage directory that is writeable to them.

rleigh-codelibre, 2018-01-10:
All very true.  Including the bioformats version and the filename in the hashed data (and anything else relevant, e.g. architecture/OS/Java version?) would make it possible for all these variants to coexist without breakage.