ome/bioformats #3059: NDPIReader & makeRGBImage last scanlines wrong

mstritt, 2018-02-01:
When using NDPIReader and converting the data to an image using a BufferdImageReader - or makeRGBImage directly - on some images (probably depending on the height) the last few pixel rows (always 8?) have value 128 instead of the correct pixel value.

The problem can be reproduced on the example image
http://downloads.openmicroscopy.org/images/Hamamatsu-NDPI/manuel/test3-FITC%202%20(485).ndpi

Code:

```
![testndpi-zoom](https://user-images.githubusercontent.com/5435937/35679876-e5739054-0758-11e8-9513-c4bcc7cc7a10.jpg)
![testndpi](https://user-images.githubusercontent.com/5435937/35679877-e59397b4-0758-11e8-8205-4e71f3a9c994.jpg)
![testndpi](https://user-images.githubusercontent.com/5435937/35679897-f5b1459c-0758-11e8-8594-908e6ea7b0aa.jpg)
import loci.formats.FormatReader;
import loci.formats.gui.AWTImageTools;
import loci.formats.in.NDPIReader;
import java.awt.image.BufferedImage;

public class TestNDPIReader {
    public static void main(String[] args) {
        // image from http://downloads.openmicroscopy.org/images/Hamamatsu-NDPI/manuel/test3-FITC%202%20(485).ndpi
        String filename = "test3-FITC 2 (485).ndpi";
        try (FormatReader r = new NDPIReader()) {
            r.setId(filename);
            byte[] bytes = r.openBytes(0);
            int last = bytes[bytes.length-1] & 0xf;   // here the last byte (blue value) is 0, which is correct
            System.out.println(last);    // 0
            // openImage results in value 128 - which is wrong. Trace down to makeRGBImage...
           // BufferedImage bi = AWTImageTools.openImage(bytes,r,r.getSizeX(),r.getSizeY());
           // BufferedImage bi = AWTImageTools.makeImage(bytes, r.getSizeX(), r.getSizeY(), 3, true, 1, false, true, false);
            BufferedImage bi = AWTImageTools.makeRGBImage(bytes, 3, r.getSizeX(), r.getSizeY(), true);
            int lastBi = bi.getRaster().getSample(bi.getWidth()-1,bi.getHeight()-1,0);
            System.out.println(lastBi);  // 128 but should be 0
            // this problem holds for the last 8 scanline pixels (samples width*(height-1..8)*3), before it's correct
        } catch (Exception e) {
            e.printStackTrace();
        }
    }
}
```
Maybe this is related to [#875](https://github.com/openmicroscopy/bioformats/issues/875) and/or [#836](https://github.com/openmicroscopy/bioformats/pull/836)

Result of the image using a BufferedImageReader (you can see the last vpart is gray)
[image]
Zoomin:
[image]

dgault, 2018-03-05:
Hi @mstritt, thank you for reporting this issue. I was able to reproduce the behaviour using the first series of the sample file at http://downloads.openmicroscopy.org/images/Hamamatsu-NDPI/manuel/test3-FITC%202%20(485).ndpi as you suggested. From my initial investigation the problem appears to lie within the NDPIReader itself rather than the AWTImageTools functions. This will require further investigation on our side before a fix can be put in place.

melissalinkert, 2020-01-31:
Confirmed that this is still a problem with #3505 included.

closed 2020-03-20 by sbesson

jfkotw, 2020-04-09:
I just tested the fix while preparing the latest release of Orbit Image Analysis which uses Bioformats 6.4.0, and it appears that the issue is still present. I verified this by running the same code that @mstritt submitted at the top of this bug report.

jfkotw, 2020-04-09:
FYI: I pushed the code including the gradle configuration to a repo here: https://github.com/jfkotw/LargeNDPIBug/blob/master/src/main/java/Issue3059.java

melissalinkert, 2020-04-09:
6.4.0 does not contain the fix for this issue, so it is completely expected that you will still see the problem when using 6.4.0.

PR #3526 (2020-03-20): JPEG service: fix off-by-one when using restart markers to build a tile's stream
Fixes #3059

Without this fix, the bytes between the last restart marker and the end of the original JPEG
stream were ignored.

Without this PR, ```showinf hamamatsu/public/manuel/test3-FITC 2 (485).ndpi``` will show a horizontal gray stripe at the bottom of the image approximately 8 pixels high.  With this PR, the same test should show no stripe.

I'd expect this to be safe for a patch release, but some configuration updates may be necessary depending upon the results of tonight's repo test.
commit 8f308755 (2020-03-20): JPEG service: fix off-by-one when using restart markers to build a tile's stream

Fixes #3059

Without this fix, the bytes between the last restart marker and the end of the original JPEG
stream were ignored.