ome/bioformats #3062: "loci.formats.FormatException: Unmatched UUID:..." when opening an ome-tiff

StephanPreibisch, 2018-02-05:
Hi, when trying to open an ome-tiff that Fiji reads just fine when using drag & drop I get this error:

loci.formats.FormatException: Unmatched UUID: urn:uuid:de9fdd81-3c05-c2a7-1ce4-8c68ef3978c6
	at loci.formats.in.OMETiffReader.initFile(OMETiffReader.java:616)
	at loci.formats.FormatReader.setId(FormatReader.java:1397)
	at loci.formats.ImageReader.setId(ImageReader.java:839)

Do you have any idea what might cause that?

Thanks a lot,
Stephan

dgault, 2018-02-05:
Hi Stephan,

For Multi-File OME-TIFF each TiffData element will have an associated UUID which will look similar to this:

```
<UUID FileName="tubhiswt_C0_TP1.ome.tif">urn:uuid:743275b7-6726-46bd-b7bb-aca3085f429a</UUID>
```

For this particular exception it appears to be the case that a UUID exists but without the FileName being provided.

rleigh-codelibre, 2018-02-07:
@StephanPreibisch Yes, this is almost certainly the cause of the error.  If you take a look at the OME-XML metadata in the file, it will have a reference to a file with this UUID, which presumably doesn't exist in the location it's trying to read it from.  Multi-file OME-TIFFs are always intended to be used as a complete set due to the metadata describing the images in all the files of the set.  If you split up the set, the metadata will be unable to reference the missing files and the dataset will be incomplete, hence the warnings as it attempts to validate the metadata in each file on initial opening.

Keeping the complete set of files together would be the obvious recommendation, if that's possible for your purposes.  Since OME-TIFF is very flexible in how planes are distributed across multiple files, we can't make any assumptions in our reader which would let it ignore missing files--you might have spanned a single image over multiple files split by timepoint or channel, for example.  Or even scattered around randomly!  This is probably not be the case for your image, but the reader has to cater for all these possibilities, which is why we require the complete set of files to be present.

If you want to use a subset of the data and you can't retain the complete set of files, then maybe consider using `bfconvert` or an equivalent tool to select the data you wish to retain.  This will ensure the metadata remains consistent for the subset of data you want to access.  But it will duplicate the data and take time, so might not be the ideal choice.

sbesson, 2018-03-14:
This issue has been addressed and has no further activity. It is therefore being closed. Thank you
for your contributions, they are much appreciated.

closed 2018-03-14 by sbesson