ome/bioformats #3095: ImageConverter/bfconvert: Exceptions on .ome.tif and .lif inputs for .png output

shiyah, 2018-03-05:
Hello,
I've been having trouble with creating a program based on bfconvert because of a fairly hard to pinpoint problem. 
I'd thought that the problem was on my end (wrong arguments sent to the writer? some odd java thing?), but I now have some more information and therefore feel more confident in creating an issue.

On .ome.tif input:
My program gives this error:
```
java.lang.IllegalArgumentException: newPosition > limit: (89593 > 89577)
	at java.base/java.nio.Buffer.createPositionException(Buffer.java:313)
	at java.base/java.nio.Buffer.position(Buffer.java:288)
	at java.base/java.nio.ByteBuffer.position(ByteBuffer.java:1079)
	at java.base/java.nio.ByteBuffer.position(ByteBuffer.java:260)
	at loci.common.NIOFileHandle.buffer(NIOFileHandle.java:605)
	at loci.common.NIOFileHandle.skipBytes(NIOFileHandle.java:451)
	at loci.common.RandomAccessInputStream.skipBytes(RandomAccessInputStream.java:597)
	at loci.formats.out.APNGWriter.setId(APNGWriter.java:191)
	at loci.formats.ImageWriter.setId(ImageWriter.java:481)
	at converter.Converter.run(Converter.java:213) <- My program
```
bfconvert gives this error instead:
```
Exception in thread "main" loci.formats.FormatException: Unmatched UUID: urn:uuid:85a5ab3e-0ead-498f-b2cb-3531ea0788ef
        at loci.formats.in.OMETiffReader.initFile(OMETiffReader.java:775)
        at loci.formats.FormatReader.setId(FormatReader.java:1397)
        at loci.formats.ImageReader.setId(ImageReader.java:839)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:385)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:884)
```

On .lif input
My program gives:
```
java.lang.IllegalArgumentException: newPosition > limit: (1163 > 1147)
	at java.base/java.nio.Buffer.createPositionException(Buffer.java:313)
	at java.base/java.nio.Buffer.position(Buffer.java:288)
	at java.base/java.nio.ByteBuffer.position(ByteBuffer.java:1079)
	at java.base/java.nio.ByteBuffer.position(ByteBuffer.java:260)
	at loci.common.NIOFileHandle.buffer(NIOFileHandle.java:605)
	at loci.common.NIOFileHandle.skipBytes(NIOFileHandle.java:451)
	at loci.common.RandomAccessInputStream.skipBytes(RandomAccessInputStream.java:597)
	at loci.formats.out.APNGWriter.setId(APNGWriter.java:191)
	at loci.formats.ImageWriter.setId(ImageWriter.java:481)
	at converter.Converter.run(Converter.java:213)
```
bfconvert gives:
```
Exception in thread "main" java.lang.IllegalArgumentException: newPosition > limit: (1163 > 1147)
        at java.base/java.nio.Buffer.createPositionException(Unknown Source)
        at java.base/java.nio.Buffer.position(Unknown Source)
        at java.base/java.nio.ByteBuffer.position(Unknown Source)
        at java.base/java.nio.ByteBuffer.position(Unknown Source)
        at loci.common.NIOFileHandle.buffer(NIOFileHandle.java:605)
        at loci.common.NIOFileHandle.skipBytes(NIOFileHandle.java:451)
        at loci.common.RandomAccessInputStream.skipBytes(RandomAccessInputStream.java:597)
        at loci.formats.out.APNGWriter.setId(APNGWriter.java:191)
        at loci.formats.ImageWriter.setId(ImageWriter.java:481)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:543)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:884)
```

I also tested .jpg and .png inputs but they worked just fine. 

I'll include the files I used for the tests just in case it's some issue with them instead (which may well be likely considering I wasn't sure where to get some example images to test with): 
[test_sources.zip](https://github.com/openmicroscopy/bioformats/files/1779817/test_sources.zip)

rleigh-codelibre, 2018-03-05:
@shiyah With regard to the OME-TIFF input, this is a multi-file OME-TIFF, but only a single file is present.  In the metadata:

```
% tiffcomment tubhiswt_C0.ome.tif
<TiffData FirstC="0" FirstT="0" FirstZ="0" IFD="0" PlaneCount="1"><UUID FileName="tubhiswt_C0.ome.tif">urn:uuid:cb8b431f-ab4e-4e54-8351-96d4eb785014</UUID></TiffData><TiffData FirstC="1" FirstT="0" FirstZ="0" IFD="0" PlaneCount="1"><UUID FileName="tubhiswt_C1.ome.tif">urn:uuid:85a5ab3e-0ead-498f-b2cb-3531ea0788ef</UUID></TiffData>
```

That unmatched UUID `urn:uuid:85a5ab3e-0ead-498f-b2cb-3531ea0788ef` is for the file `tubhiswt_C1.ome.tif` which is missing.  If you add that file, it should work just fine.

With regard to the LIF file, this works correctly for me for OME-TIFF:

```
% bfconvert PR2729_frameOrderCombinedScanTypes.lif test.ome.tif
```

but fails for PNG:

```
% ./tools/bfconvert -series 2 PR2729_frameOrderCombinedScanTypes.lif test3.png
PR2729_frameOrderCombinedScanTypes.lif
LIFReader initializing PR2729_frameOrderCombinedScanTypes.lif
Reading header
Finding image offsets
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
[Leica Image File Format] -> test3.png [Animated PNG]
Exception in thread "main" java.lang.IllegalArgumentException
        at java.nio.Buffer.position(Buffer.java:244)
        at loci.common.NIOFileHandle.buffer(NIOFileHandle.java:605)
        at loci.common.NIOFileHandle.skipBytes(NIOFileHandle.java:451)
        at loci.common.RandomAccessInputStream.skipBytes(RandomAccessInputStream.java:597)
        at loci.formats.out.APNGWriter.setId(APNGWriter.java:191)
        at loci.formats.ImageWriter.setId(ImageWriter.java:481)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:543)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:884)
```

I suspect this might be due to the PNG format not being able to store the higher dimensions, and it also happens when trying to overwrite an already existing PNG file.  Try using a pattern to split into multiple PNGs:

```
% bfconvert  PR2729_frameOrderCombinedScanTypes.lif "PR2729-%s-%c-%z-%t.png"
PR2729_frameOrderCombinedScanTypes.lif
LIFReader initializing PR2729_frameOrderCombinedScanTypes.lif
Reading header
Finding image offsets
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
Unknown Immersion value 'null' will be stored as "Other"
Unknown Correction value 'null' will be stored as "Other"
[Leica Image File Format] -> PR2729-%s-%c-%z-%t.png [Animated PNG]
        Series 0: converted 12/12 planes (100%)
        Series 1: converted 12/12 planes (100%)
        Series 2: converted 12/12 planes (100%)
        Series 3: converted 12/12 planes (100%)
```

I hope this helps.  Please let me know if you run into any other difficulties!

Regards,
Roger

shiyah, 2018-03-05:
Thank you very much for the quick reply.

Just as a short follow-up, I've been able to get both files working for the most part, but I'm still having some small problems with the .ome.tif input.

While specifying the channel to be converted works: `./bfconvert.bat tubhiswt_C.*.ome.tif test.png -stitch -channel 1`, using a general output pattern like `./bfconvert.bat tubhiswt_C.*.ome.tif test-%c.png -stitch` does not:
```
$ ./bfconvert.bat tubhiswt_C.*.ome.tif test-%c.png -stitch
Output file test-c.png exists.
Do you want to overwrite it? ([y]/n)
y
tubhiswt_C.*.ome.tif
OMETiffReader initializing C:\Users\<me>\Downloads\bftools\.\tubhiswt_C0.ome.tif
Reading IFDs
Populating metadata
[OME-TIFF] -> test-c.png [Animated PNG]
Reading IFDs
Populating metadata
Exception in thread "main" java.lang.IllegalArgumentException: newPosition > limit: (89593 > 89577)
        at java.base/java.nio.Buffer.createPositionException(Unknown Source)
        at java.base/java.nio.Buffer.position(Unknown Source)
        at java.base/java.nio.ByteBuffer.position(Unknown Source)
        at java.base/java.nio.ByteBuffer.position(Unknown Source)
        at loci.common.NIOFileHandle.buffer(NIOFileHandle.java:605)
        at loci.common.NIOFileHandle.skipBytes(NIOFileHandle.java:451)
        at loci.common.RandomAccessInputStream.skipBytes(RandomAccessInputStream.java:597)
        at loci.formats.out.APNGWriter.setId(APNGWriter.java:191)
        at loci.formats.ImageWriter.setId(ImageWriter.java:481)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:543)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:884)
```

rleigh-codelibre, 2018-03-05:
`tubhiswt_C.*.ome.tif` - you don't need a `*` wildcard here; just specifying one of the files is sufficient.  The OME-XML metadata will let the OME-TIFF reader identify all the files that are part of the set.

Under the Windows CMD shell, `%var%` is a variable expansion, so you need to use `%%` for a literal `%` sign.  Try `test-%%c.png`?  I think that should work, but can't test it myself just now.

shiyah, 2018-03-05:
Ah, there we go. Got it working both in bfconvert and my program. 

Thank you so much for the help!

closed 2018-03-05 by shiyah