ome/bioformats #3120: DeltaVisionReader doesn't support files with more than 5 channels

jwerbin, 2018-04-04:
Hi,

When opening .dv files with more than 5 wavelengths DeltavisionReader throws an exception.

I think the culprit is in DeltavisionReader.initExtraMetadata()
https://github.com/openmicroscopy/bioformats/blob/dcefe24cf8c9191fed22dc497b4dbd38b202b925/components/formats-gpl/src/loci/formats/in/DeltavisionReader.java#L596-L597

When trying to load a 6 wavelength file using the ImageJ plugin
> java.lang.reflect.InvocationTargetException
>	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
>	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:39)
>	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:25)
>	at java.lang.reflect.Method.invoke(Method.java:597)
>	at ij.Command.runPlugIn(Command.java:148)
>	at ij.Command.runCommand(Command.java:97)
>	at ij.Executer.run(Executer.java:64)
>	at java.lang.Thread.run(Thread.java:662)
> Caused by: java.lang.ArrayIndexOutOfBoundsException: 5
>	at loci.formats.in.DeltavisionReader.initExtraMetadata(DeltavisionReader.java:744)
>	at loci.formats.in.DeltavisionReader.initFile(DeltavisionReader.java:272)
>	at loci.formats.FormatReader.setId(FormatReader.java:1320)
>	at loci.plugins.in.ImportProcess.initializeFile(ImportProcess.java:482)
>	at loci.plugins.in.ImportProcess.execute(ImportProcess.java:146)
>	at loci.plugins.in.Importer.showDialogs(Importer.java:141)
>	at loci.plugins.in.Importer.run(Importer.java:79)
>	at loci.plugins.LociImporter.run(LociImporter.java:81)
>	at ij.IJ.runUserPlugIn(IJ.java:185)
>	at ij.IJ.runPlugIn(IJ.java:152)
>	at ij.IJ.runPlugIn(IJ.java:141)
>	at HandleExtraFileTypes.openImage(HandleExtraFileTypes.java:405)
>	at HandleExtraFileTypes.run(HandleExtraFileTypes.java:57)
>	at ij.IJ.runUserPlugIn(IJ.java:185)
>	at ij.IJ.runPlugIn(IJ.java:152)
>	at ij.IJ.runPlugIn(IJ.java:141)
>	at ij.io.Opener.openWithHandleExtraFileTypes(Opener.java:432)
>	at ij.io.Opener.openImage(Opener.java:304)
>	at ij.io.Opener.openImage(Opener.java:327)
>	at ij.io.Opener.open(Opener.java:140)
>	at ij.io.Opener.open(Opener.java:68)
>	at ij.plugin.Commands.run(Commands.java:27)
>	at ij.IJ.runPlugIn(IJ.java:158)

Can you enable DeltavisionReader to read .dv files with an arbitrary number of wavelengths?
Thanks,
Jeff

sbesson, 2018-04-05:
Hi Jeff,

Thanks for reporting this issue

The Deltavision format [specification](https://docs.openmicroscopy.org/bio-formats/5.8.1/formats/deltavision.html) and the [samples files](https://downloads.openmicroscopy.org/images/DV/) we have received so far are strictly limited to 5 channels. Are these 6-wavelengths image files a new feature of the acquisition software? If so, they probably are generated using a newer version of the file format.

Without supporting datasets or specification document, it is not possible to assess the work required to develop a reader for this new version of the file format. Do you have either sample data and ideally a specification that could be shared publicly? Also is GE Life Sciences aware of this issue? If an update to the Deltavision software broke your workflow, perhaps you should report it to them?

Best,
Sebastien

jwerbin, 2018-04-09:
Thanks Sebastien.

I will contact GE about the issue.
Though I am working with a file type derived from the dv format, so I specific reader for these files might be in order.
If I wanted to write this a new format reader and contribute it to the project how do I go about it?

Jeff

dgault, 2018-04-16:
Hi Jeff,

If you need to you need to create a new reader for the derived file format you are using then the first place to start would be the documentation link below explaining how to go about developing a new reader. If you need any help or have any questions with this please just let use know.
https://docs.openmicroscopy.org/bio-formats/5.8.1/developers/reader-guide.html

Before we could properly look at incorporating a new reader into the project we would need to have further information on the existing use case and future maintainability of the reader along with version information, a format specification and sample files. The easiest way to start using a new reader would be to have it as an external reader which can be used alongside Bio-Formats. One example of this external use case would be SlideBook reader which is developed and maintained by 3i https://docs.openmicroscopy.org/bio-formats/5.7.2/formats/3i-slidebook.html 

David

jmuhlich, 2018-05-22:
Hi Jeff.  :)  I was poking around in the dv/rcpnl format and bioformats reader myself recently. If I recall correctly the header is all fixed-length fields -- the designers only allocated space for 5 copies of each of the channel-related metadata fields.

melissalinkert, 2018-06-29:
Hi Jeff,

If you have at least one sample file that you are able to share, we are prepared to re-investigate this issue.  We have reason to believe that the fix is straightforward and will not require a new reader, but at least one file is needed for verification and testing.  Files can be sent to the OME team via https://www.openmicroscopy.org/qa2/qa/upload/

-Melissa

jwerbin, 2018-06-29:
Melissa,
Thanks for looking into this. 
I have uploaded a file named RC_SevenChannelScan.rcpnl via the link above.
It is a single position/time point scan with "channels" and each channel contains an identical test image.
I have uploaded a tiff of one of the test images for a sanity check (RC_SevenChannelScan-singleChannel.tif).

It would be great if you can get this to work but if you can't it may need  a separate reader with new spec. If that is the case I'd consider this issue closed.

Thanks,
Jeff

melissalinkert, 2018-07-02:
Thank you for uploading sample data.  We're working on a fix now as noted here:

https://trello.com/c/nik6LmGE/15-deltavision-with-more-than-5-channels

sbesson, 2018-08-01:
Closing as per #3181. This should be included in the Bio-Formats 5.9.1 release

closed 2018-08-01 by sbesson

mtbc, 2018-08-14:
Release testing confirms the fix.