ome/bioformats #3134: Undefined function or variable 'row' - MATLAB toolbox

kouichi-c-nakamura, 2018-04-30:
When opening an image with LUT (`isempty(colorMaps) == false`), `bfopen` of Bio-Formats MATLAB toolbox failed to work. Both variables row and col in line 163 are not defined in `bfopen`, so it appears to be a bug of Bio-Formats MATLAB toolbox (version 5.8.2).

```
Undefined function or variable 'row'.
Error in bfopen (line 163)
            m = newMap(row, col) < 0; 
```

I'm happy to upload a demo data for this if needed.

Meanwhile, I disabled the colorMaps by following:

```matlab
% line 161 of bfopen
if 0 %~isempty(colorMaps{s, i})
    newMap = colorMaps{s, i};
    m = newMap(row, col) < 0;
    newMap(m) = newMap(m) + bppMax;
    colorMaps{s, i} = newMap / (bppMax - 1);
end
```

The same issue has been raised in another repo.
https://github.com/ome/bio-formats-matlab/issues/5#issuecomment-385345494

sbesson, 2018-04-30:
Hi @kouichi-c-nakamura.

Thanks for opening this issue against Bio-Formats.

I cannot find the changes you are mentioning either in the [source code](https://github.com/openmicroscopy/bioformats/blob/v5.8.2/components/formats-gpl/matlab/bfopen.m#L161) or in the code shipped with the 5.8.2 toolbox:

```
sbesson@necromancer ~ $ wget http://downloads.openmicroscopy.org/bio-formats/5.8.2/artifacts/bfmatlab.zip
--2018-04-30 11:44:43--  http://downloads.openmicroscopy.org/bio-formats/5.8.2/artifacts/bfmatlab.zip
Resolving downloads.openmicroscopy.org... 134.36.65.239
Connecting to downloads.openmicroscopy.org|134.36.65.239|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 30080834 (29M) [application/zip]
Saving to: ‘bfmatlab.zip’

bfmatlab.zip        100%[=====================>]  28.69M  74.2MB/s   in 0.4s   

2018-04-30 11:44:44 (74.2 MB/s) - ‘bfmatlab.zip’ saved [30080834/30080834]

sbesson@necromancer ~ $ unzip bfmatlab.zip 
Archive:  bfmatlab.zip
   creating: bfmatlab/
   creating: bfmatlab/private/
  inflating: bfmatlab/bfCheckJavaMemory.m  
  inflating: bfmatlab/bfCheckJavaPath.m  
  inflating: bfmatlab/bfGetFileExtensions.m  
  inflating: bfmatlab/bfGetPlane.m   
  inflating: bfmatlab/bfGetReader.m  
  inflating: bfmatlab/bfInitLogging.m  
  inflating: bfmatlab/bfOpen3DVolume.m  
  inflating: bfmatlab/bfUpgradeCheck.m  
  inflating: bfmatlab/bfopen.m       
  inflating: bfmatlab/bfsave.m       
  inflating: bfmatlab/createMinimalOMEXMLMetadata.m  
  inflating: bfmatlab/private/is_octave.m  
  inflating: bfmatlab/bioformats_package.jar  
sbesson@necromancer ~ $ grep -n newMap bfmatlab/bfopen.m 
161:            newMap = single(colorMaps{s, i});
162:            newMap(newMap < 0) = newMap(newMap < 0) + bppMax;
163:            colorMaps{s, i} = newMap / (bppMax - 1);
```

Is it possible you have a local modified copy of this file? Could you try replacing your local folder with a fresh copy of 5.8.2 toolbox and see if this solves the issue for you?

kouichi-c-nakamura, 2018-04-30:
Sorry for bothering you, in fact, I can't find the same code any longer. It looks like the version history of my bfopen is a little messed up at some point. I think we can close this issue for now.

sbesson, 2018-04-30:
No worries @kouichi-c-nakamura. Closing this issue. Hopefully, the default file will work as expected.

closed 2018-04-30 by sbesson