ome/bioformats #3164: ND2 Fast Timelapse Timestamp Parse Incorrect

MXGHarryLiu, 2018-06-11:
Hi BioFormats developers and community members, 

When I use BioFormat 5.8.2 in ImageJ (FIJI) 1.52c or MATLAB, the timestamp for each frame of ND2 files recorded by fast timelapse mode in NIS-Elements (so as to achieve 1 kHz frame rate) show inconsistent numbers compared to the NIS-Elements Viewer. 
I have upload a sample recording on [Google drive](https://drive.google.com/open?id=1WIGcyVK8Uui2Ecx1eKN8VuzAfn_kBUWj), and a screenshot below: 
[image]

Thanks for your support, 
Harry

dgault, 2018-06-13:
Hi @MXGHarryLiu, thank you for the report. This looks as though the version of the file likely has the timepoints labelled differently than Bio-Formats expects. If you have a sample file that shows this behaviour it would be very helpful for testing. You can upload it to https://www.openmicroscopy.org/qa2/qa/upload/

MXGHarryLiu, 2018-06-13:
@dgault Thanks for the prompt reply. I have uploaded the same file as I tested to the [site](https://www.openmicroscopy.org/qa2/qa/feedback/21532/). 
Upon close inspection of the file content, the time points of fast time sequence are stored as custom data named `AcqTimesCache`, i.e. with the header of `CustomData|AcqTimesCache!` in unit of millisecond. If you want more sample files, please let me know. 
Hope this information may be helpful.

dgault, 2018-06-14:
Thanks for the info @MXGHarryLiu, I can see there are 3 instances of AcqTimesCache for the sample file provided, the first has 5000 values which are the incorrect values you are currently seeing, the second is called AcqTimesCache2 with 10000 values which all appear to be 0. The third is also called AcqTimesCache with 10000 values and this last set appears to be the correct data for the timepoints.

Do you have any idea what the other 2 instances of `AcqTimesCache` are storing?

MXGHarryLiu, 2018-06-14:
As far as multiple instances of `AcqTimesCache`, I don't know the cause other than guessing it's highly related to our equipment setup, but I can tell that BioFormats correctly infers `dMaxPeriodDiff` and `dMinPeriodDiff` as the max or min of frame intervals, maybe they can provide information for additional conditional branching and the root cause. 
In addition, I inspect other files taken with different modality, and they also have two instances of time cache: `CustomData|AcqTimesCache!` and `CustomData|AqcTimes2Cache!` (latter being all zeros), but BioFormats is able to parse the correct frame time points (from `AcqTimesCache`). You can check the file [here](https://drive.google.com/open?id=1rXwOfKvkKC_Dv4zILSTzKrZlK-tY5kXI). If you need more counterexamples from me, please let me know.

dgault, 2018-06-15:
Thanks @MXGHarryLiu, I have opened a PR to test a potential fix against the rest of our existing repository of files - https://github.com/openmicroscopy/bioformats/pull/3168

MXGHarryLiu, 2018-06-18:
@dgault I greatly appreciate your effort. I will close this issue when the fix is released. Thanks!

MXGHarryLiu, 2020-05-04:
Recently, we had a new microscopy station with NIS Elements 5.20.01, and I've confirmed that the time points of nd2 files created by Fast Time Lapse still cannot be parsed by Bio-Formats 6.5.0. These files typically has only one channel and series, but many time points with about 1 kHz frame rate, e.g. X2000 Y50 C1 Z1 T5000 S1. 
I wonder if there could be a quick workaround in Bio-Formats to load other instances of `AcqTimesCache` once the current parsed values of time points don't make sense, e.g. too large or too precise. That is, if Bio-Formats could return formated data stored under `CustomData`. 
Currently, we are using a very slow [workaround](https://github.com/joe-of-all-trades/nd2read/blob/master/nd2finfo.m) to parse for the metadata after scanning through the entire file.

dgault, 2020-05-05:
If you have sample files for this version you can upload them to https://www.openmicroscopy.org/qa2/qa/upload/ along with the expected values.

Realistically there is not likely to be a quick fix for this in the short term. It appears that different versions of NIS Elements are storing the values slightly differently. Supporting many different variations of these proprietary formats is challenging for us and currently our main focus will be looking at open next generation formats (see https://blog.openmicroscopy.org/community/file-formats/2019/06/25/formats/). For example the previous attempt to resolve this issue didn't make it into a release as it would have solved the issue for one version of the format but also broken some older versions which we test against.

dgault, 2021-05-31:
Closing this issue as it should not be fixed by the merged PR https://github.com/ome/bioformats/pull/3692 which will be included with Bio-Formats 6.7.0

closed 2021-05-31 by dgault