ome/bioformats #3166: Errors importing MetaMorph time series with ImageJ or bfconvert

Cumol, 2018-06-13:
Here are the first 10 frames of an image series captured with MetaFlour (a MetaMorph software):

https://www.dropbox.com/s/2xvxycbwvm9p85k/test-series.zip?dl=0

Using the Fiji BioFormats Plugin Importer with the "Group files with simlar names" option ticket gives me the following error:

```
(Fiji Is Just) ImageJ 2.0.0-rc-67/1.52c; Java 1.8.0_171 [64-bit]; Mac OS X 10.11.6; 297MB of 6222MB (4%)
 
java.lang.NullPointerException
	at loci.formats.in.MetamorphTiffReader.getField(MetamorphTiffReader.java:568)
	at loci.formats.in.MetamorphTiffReader.initFile(MetamorphTiffReader.java:243)
	at loci.formats.FormatReader.setId(FormatReader.java:1397)
	at loci.plugins.in.ImportProcess.initializeFile(ImportProcess.java:499)
	at loci.plugins.in.ImportProcess.execute(ImportProcess.java:142)
	at loci.plugins.in.Importer.showDialogs(Importer.java:140)
	at loci.plugins.in.Importer.run(Importer.java:76)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:222)
	at ij.IJ.runPlugIn(IJ.java:186)
	at ij.Executer.runCommand(Executer.java:137)
	at ij.Executer.run(Executer.java:66)
	at java.lang.Thread.run(Thread.java:748)
```

The same happens when using the bfconvert tool. Either on the whole folder or a singe file:

```
./bfconvert -debug ~/Desktop/test/23.001 23.001.tif
```

Gives this:

```
No update needed
/Users/MuadWork/Desktop/test/23.001
Could not find loci.formats.in.SlideBook6Reader
java.lang.ClassNotFoundException: loci.formats.in.SlideBook6Reader
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381) ~[na:1.8.0_171]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424) ~[na:1.8.0_171]
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:349) ~[na:1.8.0_171]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357) ~[na:1.8.0_171]
	at java.lang.Class.forName0(Native Method) ~[na:1.8.0_171]
	at java.lang.Class.forName(Class.java:264) ~[na:1.8.0_171]
	at loci.formats.ClassList.parseLine(ClassList.java:196) [bioformats_package.jar:5.8.2]
	at loci.formats.ClassList.parseFile(ClassList.java:258) [bioformats_package.jar:5.8.2]
	at loci.formats.ClassList.<init>(ClassList.java:138) [bioformats_package.jar:5.8.2]
	at loci.formats.ClassList.<init>(ClassList.java:122) [bioformats_package.jar:5.8.2]
	at loci.formats.ImageReader.getDefaultReaderClasses(ImageReader.java:79) [bioformats_package.jar:5.8.2]
	at loci.formats.ImageReader.<init>(ImageReader.java:116) [bioformats_package.jar:5.8.2]
	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:346) [bioformats_package.jar:5.8.2]
	at loci.formats.tools.ImageConverter.main(ImageConverter.java:884) [bioformats_package.jar:5.8.2]
Could not find loci.formats.in.ScreenReader
java.lang.ClassNotFoundException: loci.formats.in.ScreenReader
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381) ~[na:1.8.0_171]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424) ~[na:1.8.0_171]
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:349) ~[na:1.8.0_171]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357) ~[na:1.8.0_171]
	at java.lang.Class.forName0(Native Method) ~[na:1.8.0_171]
	at java.lang.Class.forName(Class.java:264) ~[na:1.8.0_171]
	at loci.formats.ClassList.parseLine(ClassList.java:196) [bioformats_package.jar:5.8.2]
	at loci.formats.ClassList.parseFile(ClassList.java:258) [bioformats_package.jar:5.8.2]
	at loci.formats.ClassList.<init>(ClassList.java:138) [bioformats_package.jar:5.8.2]
	at loci.formats.ClassList.<init>(ClassList.java:122) [bioformats_package.jar:5.8.2]
	at loci.formats.ImageReader.getDefaultReaderClasses(ImageReader.java:79) [bioformats_package.jar:5.8.2]
	at loci.formats.ImageReader.<init>(ImageReader.java:116) [bioformats_package.jar:5.8.2]
	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:346) [bioformats_package.jar:5.8.2]
	at loci.formats.tools.ImageConverter.main(ImageConverter.java:884) [bioformats_package.jar:5.8.2]
processor is INTEL_64 os.arch is x86_64
architecture is OSX_64 os.name is mac os x
architecture is OSX_64 os.name is mac os x
platform specific path is META-INF/lib/osx_64/
mappedLib is libturbojpeg.dylib
URL is jar:file:/Users/MuadWork/Downloads/bftools/bioformats_package.jar!/META-INF/lib/osx_64/libturbojpeg.dylib
URL path is file:/Users/MuadWork/Downloads/bftools/bioformats_package.jar!/META-INF/lib/osx_64/libturbojpeg.dylib
Extracting 'jar:file:/Users/MuadWork/Downloads/bftools/bioformats_package.jar!/META-INF/lib/osx_64/libturbojpeg.dylib' to '/var/folders/rs/3g07sjxx4g7_vm810yw7bp8c0000gn/T/libturbojpeg8532646881736721004.dylib'
Loaded properties from: services.properties
Added interface interface loci.formats.services.POIService and implementation class loci.formats.services.POIServiceImpl
Added interface interface loci.formats.services.MDBService and implementation class loci.formats.services.MDBServiceImpl
Added interface interface loci.formats.services.JPEGTurboService and implementation class loci.formats.services.JPEGTurboServiceImpl
Added interface interface ome.codecs.services.LuraWaveService and implementation class ome.codecs.services.LuraWaveServiceImpl
Added interface interface loci.formats.services.JAIIIOService and implementation class loci.formats.services.JAIIIOServiceImpl
Added interface interface loci.formats.services.WlzService and implementation class loci.formats.services.WlzServiceImpl
Added interface interface loci.formats.services.JHDFService and implementation class loci.formats.services.JHDFServiceImpl
Added interface interface loci.formats.services.NetCDFService and implementation class loci.formats.services.NetCDFServiceImpl
Added interface interface loci.formats.services.EXIFService and implementation class loci.formats.services.EXIFServiceImpl
Added interface interface loci.formats.services.MetakitService and implementation class loci.formats.services.MetakitServiceImpl
Added interface interface loci.formats.services.LuraWaveService and implementation class loci.formats.services.LuraWaveServiceImpl
Added interface interface loci.formats.services.OMEXMLService and implementation class loci.formats.services.OMEXMLServiceImpl
Added interface interface ome.codecs.services.JAIIIOService and implementation class ome.codecs.services.JAIIIOServiceImpl
Added interface interface loci.formats.services.JPEGXRService and implementation class loci.formats.services.JPEGXRServiceImpl
Using mapped byte buffer? false
Attempting to update XML with version: 
XML updated to at least 2008-09
Running UPDATE_200809 stylesheet.
XML updated to at least 2009-09
Running UPDATE_200909 stylesheet.
XML updated to at least 2010-04
Running UPDATE_201004 stylesheet.
XML updated to at least 2010-06
Running UPDATE_201006 stylesheet.
XML updated to at least 2011-06
Running UPDATE_201106 stylesheet.
XML updated to at least 2012-06
Running UPDATE_201206 stylesheet.
XML updated to at least 2013-06
Running UPDATE_201306 stylesheet.
XML updated to at least 2015-01
Running UPDATE_201501 stylesheet.
XML updated to at least 2016-06
MetamorphTiffReader initializing /Users/MuadWork/Desktop/test/23.001
MetamorphTiffReader initializing /Users/MuadWork/Desktop/test/23.001
loci.formats.in.MetamorphTiffReader.initFile(/Users/MuadWork/Desktop/test/23.001)
Reading IFDs
Populating metadata
Populating OME metadata
Invalid timestamp '20180612 11:56:54.672'
Invalid timestamp '20180612 11:56:54.672'
Invalid timestamp '20180612 11:56:54.672'
Expected positive value for PhysicalSize; got 0.0
Expected positive value for PhysicalSize; got 0.0
Exception in thread "main" java.lang.NullPointerException
	at loci.formats.in.MetamorphTiffReader.getField(MetamorphTiffReader.java:568)
	at loci.formats.in.MetamorphTiffReader.initFile(MetamorphTiffReader.java:243)
	at loci.formats.FormatReader.setId(FormatReader.java:1397)
	at loci.formats.ImageReader.setId(ImageReader.java:842)
	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:385)
	at loci.formats.tools.ImageConverter.main(ImageConverter.java:884)
```

dgault, 2018-06-14:
Hi @Cumol, thank you reporting this issue and providing a sample file. 

I was able to test and reproduce the NullPointerException. I have opened a PR (https://github.com/openmicroscopy/bioformats/pull/3167) with a potential fix for this problem.

Cumol, 2018-06-14:
Thanks for the quick fix. How do I get this update into Fiji? Probably have to wait until the next release?

dgault, 2018-06-15:
Yeah, I will try to aim to get this included for the next 5.9.0 release which is scheduled for the end of the month.

sbesson, 2018-06-19:
@Cumol thank you very much for reporting this issue and providing data to reproduce. In order for us to finalize the code fix , we would like to add the sample dataset you shared to our curated QA repository for non-regression testing. Would that fine from your perspective? Additionally, could this dataset be made public under a CC-BY license at http://downloads.openmicroscopy.org/images/ (otherwise we will keep it private)?

Cumol, 2018-06-19:
@sbesson yeah, no problem :)

sbesson, 2018-07-30:
Should now be fixed by 5.9.0. Closing

closed 2018-07-30 by sbesson