ome/bioformats #3186: Combine multiple metamorph files into sinlge .tif file

Cumol, 2018-07-17:
I am trying to concatenate a time series of MetaMorph files into a singe .tif file with bfconvert (command line tool) but cannot make it work.

Is it at all possible? If so, what is the correct syntax?

sbesson, 2018-07-30:
Hi @Cumol, the Bio-Formats command-line tools and more generally `IFormatReader.setId` method is expecting a file as the input and will not handle directories as suggested by the exception.

Converting Metamorph files into a single (OME-)TIFF should definitely work as long as you point the `bfconvert` utility to one of the [files consituting the fileset](https://docs.openmicroscopy.org/bio-formats/5.9.0/formats/dataset-table.html).

Using your test files and Bio-Formats 5.9.0, the two commands below should produce a single OME-TIFF and TIFF image:

```
sbesson@ls30630:Downloads $ bfconvert 23.001 23.ome.tiff
23.001
MetamorphTiffReader initializing 23.001
Reading IFDs
Populating metadata
Populating OME metadata
[Metamorph TIFF] -> 23.ome.tiff [OME-TIFF]
	Converted 1/10 planes (10%)
	Converted 10/10 planes (100%)
[done]
6.068s elapsed (3.5+325.3ms per plane, 1761ms overhead)
sbesson@ls30630:Downloads $ bfconvert 23.001 23.tiff
23.001
MetamorphTiffReader initializing 23.001
Reading IFDs
Populating metadata
Populating OME metadata
[Metamorph TIFF] -> 23.tiff [Tagged Image File Format]
	Converted 1/10 planes (10%)
	Converted 10/10 planes (100%)
[done]
6.69s elapsed (3.9+216.6ms per plane, 3615ms overhead)
```

Cumol, 2018-08-10:
@sbesson wanted to try it out and wondered why it still gives the same error. It seems like th command line tools on the website are still the older version:

https://docs.openmicroscopy.org/bio-formats/5.8.2/users/comlinetools/index.html

When I run bfconvert -version it outputs 5.6.0 (from August 2017)

Cumol, 2018-08-10:
I tried one full dataset and was able to convert it correctly.

Then tried another one and got the following output:

'''
23.001
MetamorphTiffReader initializing 23.001
Reading IFDs
Populating metadata
Populating OME metadata
Error reading IFD type at: 6
Exception in thread "main" java.lang.NullPointerException
	at loci.common.xml.XMLTools.sanitizeXML(XMLTools.java:298)
	at loci.formats.in.MetamorphTiffReader.parseFile(MetamorphTiffReader.java:628)
	at loci.formats.in.MetamorphTiffReader.initFile(MetamorphTiffReader.java:217)
	at loci.formats.FormatReader.setId(FormatReader.java:1397)
	at loci.formats.ImageReader.setId(ImageReader.java:842)
	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:385)
	at loci.formats.tools.ImageConverter.main(ImageConverter.java:884)
'''

currently trying to find out how to replicate this with a small number of files

closed 2025-07-02 by sbesson

hummuscience, 2025-07-02:
Wow! Thanks for still delivering this 6 years later <3

PR #4337 (2025-07-02): Metamorph TIFF: throw more explicit exceptions if parsing invalid TIFF
Fixes #3186.

If a TIFF without a valid IFD, or without the expected XML metadata, is encountered, this throws FormatException with the name of the affected file. That should be clearer than a NullPointerException with no indication of which file is causing the problem.

In this particular reader, each TIFF needs to have readable metadata in order to determine which specific plane (channel, Z section, etc.) the file refers to. Per-file metadata is also used to calculate the C, Z, and T sizes. That makes simply skipping prob [... cut]
commit 3011e08e (2025-07-02): Metamorph TIFF: throw more explicit exceptions if parsing invalid TIFF

Fixes #3186.

If a TIFF without a valid IFD, or without the expected XML metadata,
is encountered, this throws FormatException with the name of the
affected file. That should be clearer than a NullPointerException with
no indication of which file is causing the problem.

In this particular reader, each TIFF needs to have readable metadata
in order to determine which specific plane (channel, Z section, etc.)
the file refers to. Per-file metadata is also used to calculate the
C, Z, and T sizes. That makes simply skipping pro [... cut]