ome/bioformats #3188: Error while reading .dm4 files

adroitanup, 2018-07-26:
Is there a way to read .dm4 files using bioformats? I keep on getting an error.
[image]

dgault, 2018-07-31:
Hi @adroitanup, .dm4 files should be supported. If you have a sample file you can upload to https://www.openmicroscopy.org/qa2/qa/upload/ I would be happy to test it for you to see what is causing the problem.

melissalinkert, 2018-12-19:
Thanks again for reporting this, @adroitanup.  https://github.com/openmicroscopy/bioformats/pull/3288 seems to have fixed part of the problem; https://github.com/openmicroscopy/bioformats/pull/3294 should fix the remainder.

Closing in favor of https://github.com/openmicroscopy/bioformats/pull/3294 - we'll update there with any testing progress.

closed 2018-12-19 by melissalinkert