ome/bioformats #3199: Dimension order issues when writing ome-tiff files...

markehammons, 2018-08-15:
As per the minimal writer example, I'm using this code to populate my OME-TIFF metadata:
```
    MetadataTools.populateMetadata(meta,0, output.getName, true, "XYZCT",
      FormatTools.getPixelTypeString(pixelType), dimensionality.xDimension, dimensionality.yDimension, dimensionality.zDimension, 1, 1, 1)
```
Everything else is being recorded properly, except the dimension order is being reported by the bioformats importer included in fiji as "XYCZT".

The odd thing is that when I do meta.getPixelsDimensionOrder(0).getValue, it returns XYZCT as I wrote it initially. What's going on here? Is the dimension order of ome-tiff files set in stone to be XYCZT?

melissalinkert, 2018-08-15:
> Everything else is being recorded properly, except the dimension order is being reported by the bioformats importer included in fiji as "XYCZT".

Do you see ```XYCZT``` only in the ```Stack order``` option of the ```Bio-Formats Import Options``` window (as in http://docs.openmicroscopy.org/bio-formats/5.9.1/users/imagej/load-images.html#opening-files)?
If so, that is expected behavior.  The ```Stack order``` option indicates the order in which planes will be read by the plugin, not the order in which planes are actually stored in the file.  ImageJ's Hyperstack feature expects planes in ```XYCZT``` order, so the plugin automatically translates from whatever order is present in the file to ```XYCZT```.  If you check the ```Display OME-XML metadata``` option, you should see the original ```XYZCT``` value, which represents how the planes are actually stored in the OME-TIFF.

> The odd thing is that when I do meta.getPixelsDimensionOrder(0).getValue, it returns XYZCT as I wrote it initially. What's going on here? Is the dimension order of ome-tiff files set in stone to be XYCZT?

OME-TIFF definitely allows any dimension order indicated in the OME-XML schema, not just ```XYCZT```.  If you see an incorrect dimension order anywhere other than the ```Stack order``` option, then that's something we'll need to investigate further.

markehammons, 2018-08-16:
@melissalinkert Yeah, i just realized it was importing into fiji with xyczt order. Sorry, I'm dumb.

sbesson, 2025-06-29:
Closing as there is the expected behavior

closed 2025-06-29 by sbesson