ome/bioformats #3200: BioFormats crashes when trying to open second slice - bioformats or image problem?

bgriffen, 2018-08-16:
The images in question can be found [here](https://drive.google.com/drive/folders/19ETt4fdci-H2lkHOflP-L-XvJm9KLtW7?usp=sharing); anyone works as an example here. 

I recently was applying the [BigStitcher](https://github.com/PreibischLab/BigStitcher) plugin to stitch some tiles together but because that module uses BioFormats, it Fiji was encountering an opening error. The identification of this issue occurred [here](https://github.com/PreibischLab/BigStitcher/issues/16).

The output I get on my (up-to-date) Fiji is as follows:

```
(Fiji Is Just) ImageJ 2.0.0-rc-68/1.52e; Java 1.8.0_66 [64-bit]; Linux 4.15.0-30-generic; 117MB of 230000MB (<1%)
 
java.lang.ArrayIndexOutOfBoundsException
	at java.lang.System.arraycopy(Native Method)
	at loci.formats.tiff.TiffParser.getSamples(TiffParser.java:1087)
	at loci.formats.tiff.TiffParser.getSamples(TiffParser.java:834)
	at loci.formats.in.MinimalTiffReader.openBytes(MinimalTiffReader.java:306)
	at loci.formats.DelegateReader.openBytes(DelegateReader.java:227)
	at loci.formats.ChannelFiller.openBytes(ChannelFiller.java:156)
	at loci.formats.ChannelSeparator.openBytes(ChannelSeparator.java:227)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:348)
	at loci.formats.DimensionSwapper.openBytes(DimensionSwapper.java:249)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:348)
	at loci.formats.MinMaxCalculator.openBytes(MinMaxCalculator.java:269)
	at loci.formats.MinMaxCalculator.openBytes(MinMaxCalculator.java:260)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
	at loci.plugins.util.ImageProcessorReader.openProcessors(ImageProcessorReader.java:186)
	at loci.plugins.in.ImagePlusReader.readProcessors(ImagePlusReader.java:422)
	at loci.plugins.in.ImagePlusReader.readPlanes(ImagePlusReader.java:387)
	at loci.plugins.in.ImagePlusReader.readImage(ImagePlusReader.java:282)
	at loci.plugins.in.ImagePlusReader.readImages(ImagePlusReader.java:243)
	at loci.plugins.in.ImagePlusReader.readImages(ImagePlusReader.java:221)
	at loci.plugins.in.ImagePlusReader.openImagePlus(ImagePlusReader.java:116)
	at loci.plugins.in.Importer.readPixels(Importer.java:149)
	at loci.plugins.in.Importer.run(Importer.java:86)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:228)
	at ij.IJ.runPlugIn(IJ.java:192)
	at ij.Executer.runCommand(Executer.java:137)
	at ij.Executer.run(Executer.java:66)
	at java.lang.Thread.run(Thread.java:745)
```
The image information after Image > Show Info is as follows:

````
(Fiji Is Just) ImageJ 2.0.0-rc-68/1.52e; Java 1.8.0_66 [64-bit]; Linux 4.15.0-30-generic; 430MB of 230000MB (<1%)
Title: Image_XY01_00001_Z002_CH1.tif
Width:  20 inches (1920)
Height:  15 inches (1440)
Size:  5.3MB
Resolution:  96 pixels per inch
Pixel size: 0.0104x0.0104 inch^2
ID: -14
Bits per pixel: 16 (unsigned)
Display range: 406 - 27423
No threshold
Magnification: 0.75
ScaleToFit: false
Uncalibrated
Path: /path/to/image/Image_XY01_00001_Z002_CH1.tif
Screen location: 89,93 (1920x1200)
No overlay
No selection
```

Would anyone have an idea why BioFormats is unable to open this image? I can open it by just dragging into Fiji but I need to be able to use 3rd party plugins and when they use BioFormats, it breaks. Is there something about my image format that is breaking BioFormats or is it an edge case not considered by BioFormats. @hoerldavid mentioned "somehow, it thinks that there are 2 planes or time points and then crashes as it tries to access the second, nonexistent, slice" and recommended I open the discussion here. Thanks in advance.

joshmoore, 2018-08-17:
Some initial notes after reproducing with the downloaded data:

Bio-Formats TiffReaders detect a time-series:

```
filename = Image_XY01_00557_Z001_CH2.tif
Series count = 1
Series #0 :
	Image count = 2
	Width = 1920
	Height = 1440
	SizeZ = 1
	SizeT = 2
	SizeC = 1
	Tile size = 1920 x 3
```

`tiffinfo` shows two images:

```
  Image Width: 1920 Image Length: 1440
  Image Width: 160 Image Length: 120
```

and additionally:

```
TIFFReadCustomDirectory: Warning, Unknown field with tag 59933 (0xea1d) encountered.
```

cF https://sno.phy.queensu.ca/~phil/exiftool/standards.html

> In February 2007 Microsoft proposed a new PhotoInfo tag called "OffsetSchema" (hex. 0xEA1D, dec. 59933) in an attempt to patch a deficiency in the EXIF maker note specification (see point 1 in EXIF 2.2 section above).... 1. For this new tag to be available to a single-pass metadata reader, it must come before the maker note data (hex. 0x927C, dec. 37500). But since the EXIF/TIFF format specifies that tags must be stored in numerical order, the maker note tag (hex. 0x927C) comes before the OffsetSchema tag (hex. 0xEA1D). 2. The OffsetSchema tag will be invalidated by any software that rewrites the EXIF and moves the maker notes without properly updating the tag.

melissalinkert, 2018-08-20:
Thanks for reporting @bgriffen, and for initial investigation @joshmoore.  The problem is that there really are two images stored in each TIFF - a full resolution image and what looks like a thumbnail (though it is not identified as thumbnail according to the TIFF spec).

This ticket is similar: https://trac.openmicroscopy.org/ome/ticket/13144

An initial pull request to split the two images into separate series is here: https://github.com/openmicroscopy/bioformats/pull/3202

If BigStitcher cannot be configured to open only the first series or you need an immediate work-around, then the easiest thing to do is convert your files to remove the extra thumbnail-like images.  An ImageJ macro to do this is here:

https://github.com/openmicroscopy/bioformats/blob/develop/components/bio-formats-plugins/utils/macros/batchTiffConvert.txt

Line 27 of that macro would need to be modified to only convert the first image in the files (replace ```imageCount``` with ```1```).

closed 2019-11-08 by sbesson