ome/bioformats #3215: MetamorphReader fails opening single-z-slice, multi-position nd datasets

imagejan, 2018-09-04:
Datasets in Metamorph format (`nd` and `stk` files) saved from Visitron's VisiView software do not open correctly when they contain 1 z slice but several stage positions (=series). Instead of displaying a _Bio-Formats Series Options_ dialog, a simple text window is opened, containing the contents of the `nd` file.

A dataset (1 `nd` file, 2 `stk` files) reproducing the issue can be found publicly [here](https://github.com/fmi-basel/faim-imagej-bio-formats-tests/tree/e95cc7f60368367c4272adba3bbe3c842f86cf57/src/test/resources/ch/fmi/benchmark_v1_2018_x64y64z1c2s4t11).

The following script reproduces the issue in Fiji:

```groovy
#@ File input
import loci.plugins.BF
imps = BF.openImagePlus(input.getPath())
```

It displays the following stack trace in the script editor console:

```
loci.formats.FormatException: Invalid image number: 4 (series=0, imageCount=4)
	at loci.formats.FormatTools.checkPlaneNumber(FormatTools.java:1008)
	at loci.formats.FormatTools.checkPlaneParameters(FormatTools.java:996)
	at loci.formats.in.MinimalTiffReader.openBytes(MinimalTiffReader.java:283)
	at loci.formats.in.MetamorphReader.openBytes(MetamorphReader.java:284)
	at loci.formats.in.MetamorphReader.openBytes(MetamorphReader.java:313)
	at loci.formats.ChannelFiller.openBytes(ChannelFiller.java:156)
	at loci.formats.ChannelSeparator.openBytes(ChannelSeparator.java:227)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:348)
	at loci.formats.DimensionSwapper.openBytes(DimensionSwapper.java:249)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:348)
	at loci.formats.MinMaxCalculator.openBytes(MinMaxCalculator.java:269)
	at loci.formats.MinMaxCalculator.openBytes(MinMaxCalculator.java:260)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
	at loci.plugins.util.ImageProcessorReader.openProcessors(ImageProcessorReader.java:186)
	at loci.plugins.in.ImagePlusReader.readProcessors(ImagePlusReader.java:422)
	at loci.plugins.in.ImagePlusReader.readPlanes(ImagePlusReader.java:387)
	at loci.plugins.in.ImagePlusReader.readImage(ImagePlusReader.java:282)
	at loci.plugins.in.ImagePlusReader.readImages(ImagePlusReader.java:243)
	at loci.plugins.in.ImagePlusReader.readImages(ImagePlusReader.java:221)
	at loci.plugins.in.ImagePlusReader.openImagePlus(ImagePlusReader.java:116)
	at loci.plugins.BF.openImagePlus(BF.java:98)
	at loci.plugins.BF.openImagePlus(BF.java:77)
	at loci.plugins.BF$openImagePlus.call(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.CallSiteArray.defaultCall(CallSiteArray.java:48)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:113)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.call(AbstractCallSite.java:125)
	at Script39.run(Script39.groovy:6)
	at org.scijava.plugins.scripting.groovy.GroovyScriptEngine.eval(GroovyScriptEngine.java:303)
	at org.scijava.plugins.scripting.groovy.GroovyScriptEngine.eval(GroovyScriptEngine.java:122)
	at javax.script.AbstractScriptEngine.eval(AbstractScriptEngine.java:264)
	at org.scijava.script.ScriptModule.run(ScriptModule.java:160)
	at org.scijava.module.ModuleRunner.run(ModuleRunner.java:168)
	at org.scijava.module.ModuleRunner.call(ModuleRunner.java:127)
	at org.scijava.module.ModuleRunner.call(ModuleRunner.java:66)
	at org.scijava.thread.DefaultThreadService$3.call(DefaultThreadService.java:238)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
```

---

I created a unit test project that reproduces this and other issues with the Metamorph format:
https://github.com/fmi-basel/faim-imagej-bio-formats-tests
The relevant failing tests are:

```
[ERROR] testOpenSingleDataset[benchmark_v1_2018_x64y64z1c2s4t11](ch.fmi.VisiViewFormatTest)  Time elapsed: 0.084 s  <<< FAILURE!
java.lang.AssertionError: Read image (single dataset)
	at ch.fmi.VisiViewFormatTest.testOpenSingleDataset(VisiViewFormatTest.java:89)

[ERROR] testOpenWithOthersInFolder[benchmark_v1_2018_x64y64z1c2s4t11](ch.fmi.VisiViewFormatTest)  Time elapsed: 0.101 s  <<< FAILURE!
java.lang.AssertionError: Read image (from folder with other datasets)
	at ch.fmi.VisiViewFormatTest.testOpenWithOthersInFolder(VisiViewFormatTest.java:94)

[ERROR] testOpenSingleDataset[benchmark_v1_2018_x64y64z1c2s4t12](ch.fmi.VisiViewFormatTest)  Time elapsed: 0.117 s  <<< FAILURE!
java.lang.AssertionError: Read image (single dataset)
	at ch.fmi.VisiViewFormatTest.testOpenSingleDataset(VisiViewFormatTest.java:89)

[ERROR] testOpenWithOthersInFolder[benchmark_v1_2018_x64y64z1c2s4t12](ch.fmi.VisiViewFormatTest)  Time elapsed: 0.052 s  <<< FAILURE!
java.lang.AssertionError: Read image (from folder with other datasets)
	at ch.fmi.VisiViewFormatTest.testOpenWithOthersInFolder(VisiViewFormatTest.java:94)
```

dgault, 2018-09-05:
Thanks @imagejan for the taking the time to put together these test. I can confirm that the exception is reproducible with the public data you linked. This issue is also independent of the ongoing issues in https://github.com/openmicroscopy/bioformats/issues/3216 and https://github.com/openmicroscopy/bioformats/pull/3032

Ive started debugging the issue and it seems as though the image count is being set incorrectly at the point https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-gpl/src/loci/formats/in/MetamorphReader.java#L1239

I will test some potential fixes against other files in the repo to see if we can resolve this

melissalinkert, 2018-12-19:
Apologies for the delay in following up, @imagejan.  This is a duplicate of https://trello.com/c/YRqsSFSd/289-metamorph-invalid-image-number.  Closing this issue as progress will be updated on the Trello card, but feel free to comment there.

closed 2018-12-19 by melissalinkert