ome/bioformats #3225: MATLAB: "loci.formats.in.*Reader" is not valid.

erjel, 2018-09-18:
I would like to use the Bio-Format [MATLAB Toolbox](https://www.openmicroscopy.org/bio-formats/downloads/) for automatically convert `.nd2`to `.tif`  files with a small script. Unfortunately I could not even mange to read the files. The script:
```MATLAB
filename = './nd2file.nd2';

addpath('./bfmatlab');
[status,version] = bfCheckJavaPath()

data = bfopen(filename);
```
produces the error message
```
status =

  logical

   1

version =

    '5.1.1'

"loci.formats.in.FilePatternReader" is not valid.
"loci.formats.in.ZipReader" is not valid.
"loci.formats.in.APNGReader" is not valid.
"loci.formats.in.JPEGReader" is not valid.
"loci.formats.in.PGMReader" is not valid.
"loci.formats.in.FitsReader" is not valid.
"loci.formats.in.PCXReader" is not valid.
"loci.formats.in.GIFReader" is not valid.
"loci.formats.in.BMPReader" is not valid.
"loci.formats.in.IPLabReader" is not valid.
"loci.formats.in.IvisionReader" is not valid.
"loci.formats.in.DeltavisionReader" is not valid.
"loci.formats.in.MRCReader" is not valid.
"loci.formats.in.GatanReader" is not valid.
"loci.formats.in.GatanDM2Reader" is not valid.
"loci.formats.in.ImarisReader" is not valid.
"loci.formats.in.OpenlabRawReader" is not valid.
"loci.formats.in.OMEXMLReader" is not valid.
"loci.formats.in.LIFReader" is not valid.
"loci.formats.in.AVIReader" is not valid.
"loci.formats.in.PictReader" is not valid.
"loci.formats.in.SDTReader" is not valid.
"loci.formats.in.EPSReader" is not valid.
"loci.formats.in.SlidebookReader" is not valid.
"loci.formats.in.AliconaReader" is not valid.
"loci.formats.in.MNGReader" is not valid.
"loci.formats.in.KhorosReader" is not valid.
"loci.formats.in.VisitechReader" is not valid.
"loci.formats.in.LIMReader" is not valid.
"loci.formats.in.PSDReader" is not valid.
"loci.formats.in.InCellReader" is not valid.
"loci.formats.in.L2DReader" is not valid.
"loci.formats.in.FEIReader" is not valid.
"loci.formats.in.NAFReader" is not valid.
"loci.formats.in.MINCReader" is not valid.
"loci.formats.in.QTReader" is not valid.
"loci.formats.in.MRWReader" is not valid.
"loci.formats.in.TillVisionReader" is not valid.
"loci.formats.in.ARFReader" is not valid.
"loci.formats.in.CellomicsReader" is not valid.
"loci.formats.in.LiFlimReader" is not valid.
"loci.formats.in.TargaReader" is not valid.
"loci.formats.in.OxfordInstrumentsReader" is not valid.
"loci.formats.in.VGSAMReader" is not valid.
"loci.formats.in.HISReader" is not valid.
"loci.formats.in.WATOPReader" is not valid.
"loci.formats.in.SeikoReader" is not valid.
"loci.formats.in.TopometrixReader" is not valid.
"loci.formats.in.UBMReader" is not valid.
[... 126 lines cut ...]
But if manually check it
```
r = javaObject('loci.formats.in.ND2Reader');
r.isThisType(filename, true)
```
I get a positiv result:
```
ans =

  logical

   1
```
On the other hand a 
```
r.setId(filename)
```
will result in the same error message like above.

I thought that I have followed the [documentation](https://docs.openmicroscopy.org/bio-formats/5.9.2/users/matlab/index.html) as closely as possible. But somewhere I missed something out.
I already tested it on two systems:
MATLAB 2017b with WIndows 10 64-bit and the latest Java JDK 10 (binary directory `C:\Program Files\java\jre-10.0.2\bin` added to `PATH` environment variable)
MATLAB 2017a with Debian 9 stretch 64-bit with openjdk version "1.8.0_181"

Can anyone help me?

sbesson, 2018-09-19:
Hi @erjel,

sorry to hear about these issues. I agree that the reported version number is suspicious. To help troubleshooting this issue, could you send us the content of

```
>> javaclasspath
```

in order to confirm there is nothing in the static of the dynamic Java class path that would shadow the content of `bioformats_package` you downloaded?

erjel, 2018-09-19:
Hi @sbesson ,

```
>> javaclasspath

		STATIC JAVA PATH

	C:\Micro-Manager-2.0beta\plugins\Micro-Manager\guava-17.0.jar                                             
	C:\Program Files\MATLAB\R2017b\java\patch                                                                 
	C:\Program Files\MATLAB\R2017b\java\jarext\ant.jar                                                        
	C:\Program Files\MATLAB\R2017b\java\jarext\ant-launcher.jar                                               
	C:\Program Files\MATLAB\R2017b\java\jarext\antlr.jar                                                      
	C:\Program Files\MATLAB\R2017b\java\jarext\felix.jar                                                      
	C:\Program Files\MATLAB\R2017b\java\jarext\scr.jar                                                        
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\activation.jar                                           
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\axiom-api.jar                                            
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\axiom-impl.jar                                           
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\axis2-adb.jar                                            
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\axis2-kernel.jar                                         
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\axis2-transport-http.jar                                 
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\axis2-transport-local.jar                                
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\commons-httpclient.jar                                   
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\geronimo-stax-api.jar                                    
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\httpcore.jar                                             
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\mail.jar                                                 
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\neethi.jar                                               
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\wsdl4j.jar                                               
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\wstx-asl.jar                                             
	C:\Program Files\MATLAB\R2017b\java\jarext\axis2\XmlSchema.jar                                            
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-anim.jar                                                 
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-extension.jar                                            
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-dom.jar                                                  
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-transcoder.jar                                           
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-awt-util.jar                                             
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-bridge.jar                                               
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-gvt.jar                                                  
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-css.jar                                                  
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-parser.jar                                               
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-ext.jar                                                  
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-gui-util.jar                                             
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-svggen.jar                                               
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-codec.jar                                                
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-svg-dom.jar                                              
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-script.jar                                               
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-util.jar                                                 
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-swing.jar                                                
	C:\Program Files\MATLAB\R2017b\java\jarext\batik-xml.jar                                                  
	C:\Program Files\MATLAB\R2017b\java\jarext\bluecove-emu.jar                                               
	C:\Program Files\MATLAB\R2017b\java\jarext\bluecove.jar                                                   
	C:\Program Files\MATLAB\R2017b\java\jarext\commons-cli.jar                                                
	C:\Program Files\MATLAB\R2017b\java\jarext\commons-codec.jar                                              
	C:\Program Files\MATLAB\R2017b\java\jarext\commons-collections-generic.jar                                
	C:\Program Files\MATLAB\R2017b\java\jarext\commons-collections.jar                                        
	C:\Program Files\MATLAB\R2017b\java\jarext\commons-compress.jar                                           
	C:\Program Files\MATLAB\R2017b\java\jarext\commons-io.jar                                                 
	C:\Program Files\MATLAB\R2017b\java\jarext\commons-lang.jar                                               
	C:\Program Files\MATLAB\R2017b\java\jarext\commons-logging.jar                                            
[... cut ...]
	C:\Micro-Manager-2.0beta\plugins\Micro-Manager\slf4j-api-1.7.6.jar                                        
	C:\Micro-Manager-2.0beta\plugins\Micro-Manager\swingx-0.9.5.jar                                           
	C:\Micro-Manager-2.0beta\plugins\Micro-Manager\xalan-2.7.1.jar                                            
	C:\Micro-Manager-2.0beta\plugins\Micro-Manager\xml-apis-1.3.04.jar                                        

		DYNAMIC JAVA PATH

	F:\###\z_MATLAB_Analysis\bfmatlab\bioformats_package.jar
```

Thank you for your help!
Eric

sbesson, 2018-09-19:
Hi @erjel 

```
	C:\Micro-Manager-2.0beta\plugins\Micro-Manager\formats-api-5.1.1.jar                                      
	C:\Micro-Manager-2.0beta\plugins\Micro-Manager\formats-common-5.1.1.jar  
```

largely explains the issue you are describing. Micro-Manager 1.4 and 2.0-beta both use and ship an older version of Bio-Formats.
If you want to use a recent version of Bio-Formats for reading files, you will need to remove the Micro-Manager JARs (at least the Bio-Formats ones) from your static Java path.

sbesson, 2018-09-19:
The above is actually related to the same problem. The Bio-Formats 5.1.1 JARs shipped with Micro-Manager depend on the `ome-xml:5.1.1`component which is an older version of the OME model. Removing either:

```
	C:\Micro-Manager-2.0beta\plugins\Micro-Manager\ome-xml-5.1.1.jar                                          
```

should allow to import the latest classes included in `bioformats_package`.

erjel, 2018-09-19:
Thank you a lot! It solved the problem.

**Summary:**
1.) Inspect the `javaclasspath` in matlab
````
>> javaclasspath
```` 
2.) In case multiple versions of bioformats are present comment/ delete them. 
```
>> cd(prefdir)
>> edit javaclasspath.txt
```
In my case we changed corresponding lines to:
```
[...]
#C:\Micro-Manager-2.0beta\plugins\Micro-Manager\formats-api-5.1.1.jar
#C:\Micro-Manager-2.0beta\plugins\Micro-Manager\formats-common-5.1.1.jar
[...]
#C:\Micro-Manager-2.0beta\plugins\Micro-Manager\ome-xml-5.1.1.jar
[...]
```

closed 2018-09-19 by erjel