ome/bioformats #3234: Error when trying to read CZI with BioFormats 5.9.2

sebi06, 2018-09-26:
Hi guys,

I tried to open a CZI (which works in ZEN) inside Fiji using BioFormats 5.9.2 and got an strange error. Without being the expet here, why throws this one a error inside NikonReader? Usually BioFormats is using the CZIReader. Any ideas?

```
(Fiji Is Just) ImageJ 2.0.0-rc-68/1.52g; Java 1.8.0_172 [64-bit]; Windows 7 6.1; 99MB of 48987MB (<1%)
 
java.lang.NoSuchFieldError: mergeSubIFDs
	at loci.formats.in.NikonReader.<init>(NikonReader.java:119)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at java.lang.Class.newInstance(Class.java:442)
	at loci.formats.ImageReader.<init>(ImageReader.java:129)
	at loci.formats.in.FilePatternReader.<init>(FilePatternReader.java:77)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at java.lang.Class.newInstance(Class.java:442)
	at loci.formats.ImageReader.<init>(ImageReader.java:129)
	at loci.plugins.util.LociPrefs.makeImageReader(LociPrefs.java:100)
	at loci.plugins.in.ImportProcess.createBaseReader(ImportProcess.java:619)
	at loci.plugins.in.ImportProcess.initializeReader(ImportProcess.java:485)
	at loci.plugins.in.ImportProcess.execute(ImportProcess.java:138)
	at loci.plugins.in.Importer.showDialogs(Importer.java:140)
	at loci.plugins.in.Importer.run(Importer.java:76)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:228)
	at ij.IJ.runPlugIn(IJ.java:192)
	at ij.IJ.runPlugIn(IJ.java:181)
	at HandleExtraFileTypes.openImage(HandleExtraFileTypes.java:499)
	at HandleExtraFileTypes.run(HandleExtraFileTypes.java:72)
	at ij.IJ.runUserPlugIn(IJ.java:228)
	at ij.IJ.runPlugIn(IJ.java:192)
	at ij.IJ.runPlugIn(IJ.java:181)
	at ij.io.Opener.openWithHandleExtraFileTypes(Opener.java:516)
	at ij.io.Opener.openImage(Opener.java:370)
	at ij.io.Opener.openImage(Opener.java:242)
	at ij.io.Opener.open(Opener.java:109)
	at ij.io.Opener.openAndAddToRecent(Opener.java:291)
	at ij.plugin.DragAndDrop.openFile(DragAndDrop.java:192)
	at ij.plugin.DragAndDrop.run(DragAndDrop.java:159)
	at java.lang.Thread.run(Thread.java:748)

```
Attached

[ct_interior_multichannel_small.zip](https://github.com/openmicroscopy/bioformats/files/2420512/ct_interior_multichannel_small.zip)

 CZI is here

sbesson, 2018-09-26:
Hi @sebi06, sorry to hear about this issue. There is a known issue related to an update site shipping Bio-Formats JARs incompatible with the ones maintained by the OME team and distributed via the Java 8 update site - see https://list.nih.gov/cgi-bin/wa.exe?A2=ind1808&L=IMAGEJ&F=&S=&P=76150.

Can you check which version of the `formats-bsd` JAR in your Fiji distribution? Also can you give us the list of your update sites?

sebi06, 2018-09-27:
It shows formats-bsd-5.7.3.jar and Fiji says everything is up-to-date.

[image]

And I attched the db.xml.gz or is there a more convinent way to get a lit of all activated update sites?
[db.xml.gz](https://github.com/openmicroscopy/bioformats/files/2422683/db.xml.gz)

sbesson, 2018-09-27:
Thanks @sebi06 , `formats-bsd-5.7.3.jar` is indeed the culprit. From your `db.xml.gz` file, I see you have the `MPIBPC` update site enabled. Can you try disabling it? This should upgrade `formats-bsd` to the 5.9.2 version shipped by the Java 8 update site and resolve your issue.