ome/bioformats #3245: Provide support for iiif images in Bioformats

imoutsatsos, 2018-10-11:
The international image interoperability ([iiif](https://iiif.io/about/)) framework is a widely adopted and open source supported image format. Multiple iiif compatible deep zoom image viewers and servers are developed by open source communities in the cultural heritage (read museums) and library domains. These tools are now finding increasing adoption in the biological imaging space. As a result, it would be useful if bioformats was to also support reading images and metadata from iiif servers. The iiif  Image API is described here: https://iiif.io/api/image/2.1/

Thank you for your consideration and best regards
Ioannis

ebremer, 2019-05-08:
@imoutsatsos I have built a IIIF tiling server named ImageBox based on Bioformats.  I will be releasing it to the public in a couple of months after some additional development and testing.  It was very much developed with whole slide imaging in mind.

ebremer, 2019-05-09:
ImageBox was first mentioned in this paper: https://www.ncbi.nlm.nih.gov/pmc/articles/PMC5333212/

ebremer, 2019-08-22:
Here's the working demo page that will be submitted with the ImageBox2 paper:
http://imagebox.ebremer.com

ebremer, 2019-09-17:
My paper was submitted and the code is now available.  The code is alpha and some additional features need to be implemented for full IIIF compliance, but it works.  The one unusual feature of ImageBox is that it can provide tiles over HTTP using HTTP range requests (as well as files local to the ImageBox2 instance).  The demo server runs on a non-cloud server server and pulls tiles from a WSI SVS image located on the Amazon S3 cloud.  It is fairly performant I believe.  I will be doing additional work over the next march of months to make it fully robust, but appreciate any feedback and/or comments.

joshmoore, 2019-09-25:
@ebremer. Going slightly beyond my ❤️ , I've been wondering what the next best step would be here, since this issue really isn't needed any more. What do you think about posting under https://forum.image.sc/c/announcements with the tags you think are appropriate, and let's see if we can't kick off some uptake?

All the best,
~Josh

closed 2019-09-25 by joshmoore

sbesson, 2019-09-25:
Seconding @joshmoore's suggestion to posting on image.sc to promote your work. Looking forward to reading your paper about ImageBox2. Is it already available on biorxiv or similar preprint mechanism?

ebremer, 2019-09-27:
@joshmoore @sbesson I would be happy to post something on https://forum.image.sc/c/announcements.  Let me write up some more documentation for early adopters.  It's not available on any pre-print mechanism yet but I will be happy to share the URL when it is available.