ome/bioformats #3254: Colormaps extraction inconsistent in bfopen for MATLAB

jamesmyatt, 2018-10-23:
In https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-gpl/matlab/bfopen.m, there are multiple problems relating to the reading of colormaps.

In particular:
* [line 144](https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-gpl/matlab/bfopen.m#L144): Initialises the `colorMaps` variable to be `numImages`-by-`numImages`, which doesn't make sense; it should be `numImages-by-1` (following example of `imageList`)
* [line 152 onwards](https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-gpl/matlab/bfopen.m#L152): stores the color maps in the row of `colorMaps` corresponding to the current series, leaving the other rows of `colorMaps` unused.
* [line 222](https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-gpl/matlab/bfopen.m#L222): stores the entire `colorMaps` variable for each series, when only the row with index `s` is actually used.

I can create a pull request to fix these and make it consistent with `imageList`, but it will break backwards compatibility. Is this OK?

dgault, 2018-10-25:
Line 144 does indeed look like a bug and should be changed to `numImages-by-1` as you suggested. There does not appear to be any need to keep colorMaps for each series and as such the `s` index can be dropped to bring it inline with `imageList` as recommended.

Normally we would not encourage breaking backwards compatibility but it is actually a good time to open any breaking changes for consideration. We are looking at a number of larger breaking changes for an upcoming major release at the moment, so these changes could be scheduled at the same time.

jamesmyatt, 2018-10-25:
I'm happy to implement this as long as you let me know which target branch to use

dgault, 2018-11-30:
Thanks @jamesmyatt, its always great to receive new contributions. If you wish to open a pull request you can do so against the `develop` branch

sbesson, 2025-03-17:
This

closed 2025-03-17 by sbesson