ome/bioformats #3323: CZI 8bit brightfield tiles stitched in Zen open incorrectly

Svidro, 2019-02-14:
A bit depth of 14 is standard for CZI images, and I have no problems opening original tiled images, or fused/unfused (according to Zen) stitched results at that bit depth.
When saving images in 8bit RGB, single images open just fine. Tiles open just fine as long as they have NOT been stitched. The problem arises when attempting to open stitched 8bit CZI images. All of the information is there, as shown if I uncheck stitching within the BioFormats CZI reader (FIJI) and can see each tile just fine. Selecting the "Fuse tiles" option within Zen does not seem to impact the result. With stitching checked, the image opens as shown on the left:
[image]
I would like to be able to read the 8bit stitched images for analysis purposes :)

Example files can be found here: https://drive.google.com/open?id=1QLl0nzGPzy2cxAhNIgwPCI5xGk3sDW5i
There are 3 sets of files, two 8bit examples, and one 14bit example (that opens normally). Each has an unstitched (original) version, and at least one stitched version, either fused or unfused.

Please let me know if I can provide any additional information!

dgault, 2019-02-21:
Hi @Svidro, thank you for raising the issue and providing the sample filesets. Apologies for the delayed response, the Bio-Formats 6.0.0 release has taken up a lot of time over the past week. I have been able to test and reproduce the problem as you described using this latest Bio-Formats release.

I have created a Trello card on the Bio-Formats inbox to track this problem: https://trello.com/c/sPGdYLaz/331-czi-fused-8-bit-tiles-skewed

Svidro, 2019-02-21:
Thanks for taking a look!