ome/bioformats #3324: Can't load nd2 (Nikon) illumination sequence file in MATLAB

sagarsetru, 2019-02-15:
I'm trying to load a file I acquired via Nikon illumination sequence, but I cannot load the full file. The file has 24024 frames in it and 4 channels, and is not arranged into stacks. In Elements, it opens as a time series of 24024 individual frames, as it should. But, in MATLAB, when I run:

>> % create reader for this file
>> reader = bfGetReader(nd2File);
 %get total number of planes
>> nPlanes = reader.getImageCount;

I get 4436 frames, not 24024. The reader is not picking up all the frames. I know the other frames are not in the reader, because:

>> nPlanes
nPlanes =
        4436

>> indPlane = 4437; 
>> img = double(bfGetPlane(reader, indPlane))';
Error using bfGetPlane (line 61)
The value of 'iPlane' is invalid. It must satisfy the function: @(x)isscalar(x)&&ismember(x,1:r.getImageCount()).

I was initially getting this warning:
Warning: *** Insufficient memory detected. ***
*** 352m found ***
*** 512m or greater is recommended ***
*** See http://www.mathworks.com/matlabcentral/answers/92813 ***
*** for instructions on increasing memory allocation. ***

I increased the memory in JVM, and no longer get this warning, but I still cannot load the full image from the reader. It still gives 4436 when I do reader.getImageCount, not 24024. I know it has 24024 frames because I can open it in NIS elements.

bfopen also only gives 4436 frames.

I ran bfUpgradeCheck:
>> bfUpgradeCheck
*** bioformats_package.jar is up-to-date ***

I'm using MATLAB R2017b.

Any suggestions?

dgault, 2019-02-20:
Hi @sagarsetru, apologies for the delayed response, the release of Bio-Formats 6.0.0 has taken a lot of time this week. Is the issue only happening with a single file or is it easily reproduced?

Also if you deselect the ND2 option for using the native chunkmap does that change the result? See https://docs.openmicroscopy.org/bio-formats/5.7.1/formats/options.html for the option to modify.

sagarsetru, 2019-02-20:
The updated Bio-formats seems to have resolved the issue!

Sagar

sbesson, 2019-02-21:
Thanks @sagarsetru for updating this.

closed 2019-02-21 by sbesson