ome/bioformats #3343: showinf crashing on TIFs created from VSI images by bfconvert 6.0.1

ebremer, 2019-04-09:
Using:
bfconvert -series 6 -pyramid-scale 2 -pyramid-resolutions 4 image.vsi image.tif
runs without error.

Using:
showinf image.tif results in
Exception in thread "main" java.lang.NullPointerException
        at loci.formats.in.APLReader.isThisType(APLReader.java:98)
        at loci.formats.ImageReader.getReader(ImageReader.java:188)
        at loci.formats.ImageReader.getFormat(ImageReader.java:156)
        at loci.formats.tools.ImageInfo.configureReaderPreInit(ImageInfo.java:433)
        at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1030)
        at loci.formats.tools.ImageInfo.main(ImageInfo.java:1121)

GIMP can read the TIF and show the pyramid.

joshmoore, 2019-04-10:
Hi @ebremer. Sorry for the troubles. This is a known problem that should be fixed in the next release. If you'd like to workaround, passing `-format CellSens` to `showinf` will likely help. If that doesn't, calling `showinf` with an absolute path may: `showinf /my/data/image.vsi`

You can take a look at https://github.com/openmicroscopy/bioformats/pull/3010 and the linked PRs if you'd like more information.

Cheers,
~Josh

ebremer, 2019-04-10:
I'm not running the showinf on the image.vsi but the bfconverted image.tif.  The problem did not occur when I used 6.0.0-m4.

joshmoore, 2019-04-10:
> I'm not running the showinf on the image.vsi but the bfconverted image.tif.

Understood. Then replace vsi with tif in my comments. The underlying problem here is that in trying to detect the format of the file given (vsi or tif or something else) one of the readers is misbehaving.

>  The problem did not occur when I used 6.0.0-m4.

Cheers.

ebremer, 2019-04-10:
Unfortunately, neither helped.  The error is different though.

D:\vsi>showinf -format CellSens d:\vsi\him4.tif
Checking CellSens VSI format [no]
Initializing reader
Exception in thread "main" java.lang.StringIndexOutOfBoundsException: String index out of range: -2
        at java.lang.String.substring(Unknown Source)
        at loci.formats.in.CellSensReader.initFile(CellSensReader.java:630)
        at loci.formats.FormatReader.setId(FormatReader.java:1395)
        at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
        at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1035)
        at loci.formats.tools.ImageInfo.main(ImageInfo.java:1121)

joshmoore, 2019-04-10:
> Unfortunately, neither helped.

```
D:\vsi>showinf d:\vsi\him4.tif
```
still shows the NullPointerException?

> D:\vsi>showinf -format CellSens d:\vsi\him4.tif

Sorry, this then needs to have the format updated as well:

```
showinf -format OMETiff d:\vsi\him4.tif
```

ebremer, 2019-04-10:
showinf -format Tiff d:\vsi\him4.tif worked.

NHPatterson, 2019-04-17:
I'd like to convert some ome.tiffs using ```bfconvert``` to a pyramid format but I get the same null pointer exception. Using ```showinf```` with ```-format OMEtiff```` works. Is there a similar bypass for ```bfconvert```?

petebankhead, 2019-04-27:
May this be because there is no `parent == null` check at the end of the `try` block below? For me, all TIFF images one directory below the root fail with the same error - but work if the image is moved up or down a directory, or I create a TiffReader rather than an ImageReader.
https://github.com/openmicroscopy/bioformats/blob/f9f8f17d5037dab717cb22507ef9b0d591ed9a8d/components/formats-gpl/src/loci/formats/in/APLReader.java#L85-L102

I think an explicit check for nulls here to avoid the overhead of throwing exceptions would be helpful, even if the immediate problem goes away with the changes to be more defensive against reader errors.

dgault, 2019-05-02:
There was an original PR https://github.com/openmicroscopy/bioformats/pull/2914 which was for updating the reader code to perform things like null checks. It may be worth revisiting with some tweaks to account for the changes made in the mean time, at least it will help highlight the list of readers affected.

closed 2019-05-10 by sbesson