ome/bioformats #3345: bfconvert fails on HCS fake

joshmoore, 2019-04-11:
```
/tmp/fake $bfconvert -option ometiff.companion test.companion.ome 'a&sizeC=3&plates=2&rows=2&columns=2&fields=2.fake' test_c%c.ome.tiff
a&sizeC=3&plates=2&rows=2&columns=2&fields=2.fake
FakeReader initializing a&sizeC=3&plates=2&rows=2&columns=2&fields=2.fake
[Simulated data] -> test_c%c.ome.tiff [OME-TIFF]
	Series 0: converted 3/3 planes (100%)
	Series 1: converted 3/3 planes (100%)
	Series 2: converted 3/3 planes (100%)
Exception in thread "main" loci.formats.FormatException: Plane index:3 must be < 3
	at loci.formats.FormatWriter.checkParams(FormatWriter.java:387)
	at loci.formats.out.TiffWriter.saveBytes(TiffWriter.java:221)
	at loci.formats.out.OMETiffWriter.saveBytes(OMETiffWriter.java:218)
	at loci.formats.out.OMETiffWriter.saveBytes(OMETiffWriter.java:208)
	at loci.formats.FormatWriter.saveBytes(FormatWriter.java:123)
	at loci.formats.ImageWriter.saveBytes(ImageWriter.java:246)
	at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:644)
	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:563)
	at loci.formats.tools.ImageConverter.main(ImageConverter.java:884)
```

melissalinkert, 2020-01-08:
This was reproducible as late as 6.3.0, but seems to have been fixed in 6.3.1 (likely by https://github.com/ome/bioformats/pull/3477).  Note that ```rows``` and ```columns``` are not valid .fake keys (use ```plateRows``` and ```plateCols``` instead), so the test case as written above will result in 4 series, not 16 as I assumed at first glance.

closed 2020-01-08 by melissalinkert

joshmoore, 2020-01-09:
>  Note that rows and columns are not valid .fake keys

Doh. I make that mistake all the time. Here's to synonyms or failing on stupid keys ;)

But :+1: