ome/bioformats #3357: Issue with using Bio-Formats inside Docker Container

sebi06, 2019-04-24:
HI guys,

I already asked around at the image.sc forum. See this post here for details: https://forum.image.sc/t/strange-issue-with-using-bioformats-from-python-script-inside-docker/24937/2?u=sebi06

But so far nobody seems to have an idea what is going on. In short:

- I have a python script in Fiji that saves an image as OME-TIFF using Bio-Formats, **which works fine when used from Fiji directly**
- When I use the same code inside a docker container (with the latest Fiji) I get an NullPointerException
- the "critical line of code is this one:

```
IJ.run(imps[ch], "Bio-Formats Exporter", "save=[" + savepath + "] compression=Uncompressed")
```

I am not sure if this is a bioformats issues directly, but any help or hint is really appreciated.

Sebi

```
C:\Users\m1srh\Documents\Apeer_Modules\channelsplitter_imagej_bbd81b50-c50d-4da0-ad50-47ae083c2966>docker run -it --rm -v c:\Temp\input:/input -v c:\Temp\output:/output --env-file wfe.env sebi06/ca_test_channelsplit:latest
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option PermSize=128m; support was removed in 8.0
Java HotSpot(TM) 64-Bit Server VM warning: Using incremental CMS is deprecated and will likely be removed in a future release
[INFO] Overriding BIOP Run Macro...; identifier: command:ch.epfl.biop.macrorunner.B_Run_Macro; jar: file:/Fiji.app/plugins/BIOP/B_Run_Macro-1.0.0-SNAPSHOT.jar
[INFO] Overriding Get Spine From Circle Rois; identifier: command:Cirlces_Based_Spine; jar: file:/Fiji.app/plugins/Max_Inscribed_Circles-1.1.0.jar
[INFO] Script Directory: /Fiji.app/scripts
[INFO] Opening Image: /input/3CH_Z-Stack_RN.czi
[INFO] Number of Channels to split: 3
[INFO] Done with splitting channels.
[INFO] Index ch: 0
[INFO] savepath: /output/3CH_Z-Stack_RN_C1.ome.tiff
java.lang.NullPointerException
        at loci.plugins.out.Exporter.run(Exporter.java:332)
        at loci.plugins.LociExporter.run(LociExporter.java:75)
        at ij.plugin.filter.PlugInFilterRunner.processOneImage(PlugInFilterRunner.java:266)
        at ij.plugin.filter.PlugInFilterRunner.<init>(PlugInFilterRunner.java:114)
        at ij.IJ.runUserPlugIn(IJ.java:231)
        at ij.IJ.runPlugIn(IJ.java:193)
        at ij.Executer.runCommand(Executer.java:137)
        at ij.Executer.run(Executer.java:66)
        at ij.IJ.run(IJ.java:309)
        at ij.IJ.run(IJ.java:373)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
        at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:188)
        at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:206)
        at org.python.core.PyObject.__call__(PyObject.java:515)
        at org.python.core.PyObject.__call__(PyObject.java:519)
        at org.python.pycode._pyx0.f$0(/Fiji.app/scripts/channelsplit_test.py:39)
        at org.python.pycode._pyx0.call_function(/Fiji.app/scripts/channelsplit_test.py)
        at org.python.core.PyTableCode.call(PyTableCode.java:171)
        at org.python.core.PyCode.call(PyCode.java:18)
        at org.python.core.Py.runCode(Py.java:1614)
        at org.python.core.__builtin__.eval(__builtin__.java:497)
        at org.python.core.__builtin__.eval(__builtin__.java:501)
        at org.python.util.PythonInterpreter.eval(PythonInterpreter.java:259)
        at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:57)
        at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:31)
        at javax.script.AbstractScriptEngine.eval(AbstractScriptEngine.java:264)
        at org.scijava.script.ScriptModule.run(ScriptModule.java:160)
        at org.scijava.module.ModuleRunner.run(ModuleRunner.java:168)
        at org.scijava.module.ModuleRunner.call(ModuleRunner.java:127)
        at org.scijava.module.ModuleRunner.call(ModuleRunner.java:66)
        at org.scijava.thread.DefaultThreadService$3.call(DefaultThreadService.java:238)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
```

sbesson, 2019-04-29:
I assume this is fixed by the solution proposed in https://forum.image.sc/t/strange-issue-with-using-bioformats-from-python-script-in-headless-mode-only/24937/4 i.e. enabling `windowless` mode. Closing. Let us know if this issue persists for another reason.

closed 2019-04-29 by sbesson

sebi06, 2019-04-30:
Obviously you are correct. The issues is fixed by: https://forum.image.sc/t/strange-issue-with-using-bioformats-from-python-script-in-headless-mode-only/24937/4.