ome/bioformats #3363: Read a TIFF image with 4 16bits channels

FiReTiTi, 2019-05-01:
Hi,
I use BioFormat to read 1600x1600 TIFF images containing 4 channels with 16bits encoding.
Everything was working really well, until I download the most recent version of BioFormat. With the new version, BioFormat finds only 1 channel with the image. See my outputs:

Old version:
`2019/5/1 14:49:35 - Reading image (Bio-Format) '3091443.tif' (1600x1600 <=> 2560000 pixels, 4 channels encoded with DataBufferUShort on 16 bits, type USHORT_GRAY)... successfully.`

New version:
`2019/5/1 14:31:47 - Reading image (Bio-Format) '3091443.tif'Reading IFDs`
`Populating metadata`
`Checking comment style`
`Populating OME metadata`
` (1600x1600 <=> 2560000 pixels, 1 channel encoded with DataBufferUShort on 16 bits, type USHORT_GRAY)... converting to type TYPE_USHORT_GRAY... successfully.`

The code is exactly the same, I just replaced the old version by the new one. We can see that the number of channel is not the same.

dgault, 2019-05-02:
Hi @FiReTiTi, would you be able to send us the sample file for testing? It can be uploaded through https://www.openmicroscopy.org/qa2/qa/upload/

FiReTiTi, 2019-05-02:
There was an issue with the upload system, a flash player problem (what a surprise!).
So I've uploaded one image here: www.thibault.biz/3091443.tif

sbesson, 2019-05-03:
Hi @FiReTiTi, I tested your sample file using the latest released command line tools:

```
sbesson@ls30630:tmp $ wget http://www.thibault.biz/3091443.tif
--2019-05-03 14:28:52--  http://www.thibault.biz/3091443.tif
Resolving www.thibault.biz (www.thibault.biz)... 51.68.50.220
Connecting to www.thibault.biz (www.thibault.biz)|51.68.50.220|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://www.thibault.biz/3091443.tif [following]
--2019-05-03 14:28:52--  https://www.thibault.biz/3091443.tif
Connecting to www.thibault.biz (www.thibault.biz)|51.68.50.220|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 22873775 (22M) [image/tiff]
Saving to: ‘3091443.tif’

3091443.tif         100%[===================>]  21.81M  8.86MB/s    in 2.5s    

2019-05-03 14:28:55 (8.86 MB/s) - ‘3091443.tif’ saved [22873775/22873775]

sbesson@ls30630:tmp $ showinf -version
Version: 6.0.1
Build date: 11 April 2019
VCS revision: v6.0.1
sbesson@ls30630:tmp $ showinf -nopix 3091443.tif 
Checking file format [OME-TIFF]
Initializing reader
OMETiffReader initializing 3091443.tif
Reading IFDs
Populating metadata
Initialization took 0.137s

Reading core metadata
filename = /private/tmp/3091443.tif
Used files = [/private/tmp/3091443.tif]
Series count = 1
Series #0 :
	Image count = 4
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 1600
	Height = 1600
	SizeZ = 1
	SizeT = 1
	SizeC = 4
	Tile size = 128 x 128
	Thumbnail size = 128 x 128
	Endianness = motorola (big)
	Dimension order = XYZCT (certain)
	Pixel type = uint16
	Valid bits per pixel = 16
	Metadata complete = true
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0
	Plane #2 <=> Z 0, C 2, T 0
	Plane #3 <=> Z 0, C 3, T 0

Reading global metadata

Reading metadata
```

This seems to indicate the data is correctly read as a 4-channel image using the OME metadata. Is the code you are using to call Bio-Formats available publicly?

FiReTiTi, 2019-05-03:
Hi,
no, I use it from Java:
`ImporterOptions io = new ImporterOptions() ;`
`io.setWindowless(true) ;`
`io.setQuiet(true) ;`
`io.setId(FileName) ;`
`io.setShowMetadata(false) ;`
`ImagePlus[] images = BF.openImagePlus(io) ;`
`final int channel = images[0].getNChannels() ;`
This code works fine with an older version, but not the most recent one.

dgault, 2019-05-07:
Hi @FiReTiTi, do you know which version of Bio-Formats you are using, are you working from the package jar?

FiReTiTi, 2019-05-08:
Yes, the package jar I downloaded on the website.

dgault, 2019-05-08:
Im having a hard time reproducing the issue, using the same code sample as you provided with the sample file I am seeing 4 channels ok. What values are you seeing for `images[0].getNFrames() ;` and `images[0].getNSlices() ;`?

FiReTiTi, 2019-05-08:
Mmm... the version I downloaded two weeks ago says 1 channel, 4 frames, 1 slice, and some unusual messages in the middle.
So I downloaded again the most recent version, and now I have 4 channels, 1 frame, 1 slice, and no more messages.
So I don't what happened with the version I had two weeks ago, but the problem is gone now.
Thank you for your help.

closed 2019-05-08 by FiReTiTi