ome/bioformats #3365: are images re-compressed using bfconvert

ebremer, 2019-05-07:
If I have a VSI image that is tiled using 256x256 tiles using jpg compression and I use bfconvert to convert it to a tiled bigtiff of the same tiled size and specify the same jpg compression, will the image tiles be simply copied over or are they re-compressed?

dgault, 2019-05-09:
Hi @ebremer, Im afraid in this case doing the conversion will mean that the image will be decompressed when read and then recompressed when being written to the new format.

ebremer, 2019-05-10:
Do you think it would be a monumental effort to have a conversion method that would keep the original data (assuming same tile size) and just copy it over to the new format/tile without recoding?

closed 2019-05-20 by ebremer