ome/bioformats #3370: setting JPEG compression quality with bfconvert

ebremer, 2019-05-14:
Is there a undocumented command line option to set the JPEG compression quality with bfconvert?

dgault, 2019-05-16:
Hi @ebremer, I'm afraid not, the list of compression options are limited to the below:

```
 * JPEG-2000: JPEG-2000 Lossy, JPEG-2000
 * OME-TIFF: Uncompressed, LZW, JPEG-2000, JPEG-2000 Lossy, JPEG, zlib
 * OME-XML: Uncompressed, zlib
 * QuickTime: Uncompressed
 * Tagged Image File Format: Uncompressed, LZW, JPEG-2000, JPEG-2000 Lossy, JPEG, zlib
```

ebremer, 2019-05-16:
@dgault not the types of compression, but the sub-option of JPEG which sets the compression level, usually 0-100.  I've found this in the source
CodecOptions opt = new CodecOptions();
    opt.quality = 100.0;
but it doesn't seem accessible from bfconvert. In testing this, I uncovered: https://github.com/openmicroscopy/bioformats/issues/3372

mtbc, 2019-07-02:
Not that this is also sought on [[image.sc] Bfconvert specify JPEG quality](https://forum.image.sc/t/27176).

dgault, 2022-08-04:
A further request for this feature has been received on thread https://forum.image.sc/t/error-converting-large-images-to-ome-tiff-using-the-command-line-bioformats-tool-bfconvert/70243/6

PR #4171 (2024-04-15): Add `-quality` option to bfconvert
Fixes #3370.

Builds on work in https://github.com/ome/ome-codecs/pull/26 to pass a quality value (0.0 to 1.0) through to the JPEG codec. This can be tested by comparing conversions with different quality values, e.g.:

```
$ bfconvert -compression JPEG -quality 0.1 test.fake test_10percent.tiff
$ bfconvert -compression JPEG -quality 0.5 test.fake test_50percent.tiff
$ bfconvert -compression JPEG -quality 1.0 test.fake test_100percent.tiff
```

and different output formats (TIFF, OME-TIFF, DICOM). The output file size should increase as the quality increases. [... cut]
commit 303b3d58 (2024-04-15): Add `-quality` option to bfconvert

This primarily affects JPEG compression with TIFF, OME-TIFF, and DICOM output.

Fixes #3370.
closed 2024-04-15 by dgault