ome/bioformats #3379: Strange error message when using LOCI reader inside script

sebi06, 2019-05-22:
HI guys,

I am using the LOCI reader inside a python script (latest Fiji from 20150522) and get:

ERROR loci.formats.ImageReader -  *** One or more readers is misbehaving. See the debug output for more information. e.g.: loci.formats.in.APLReader@44c82053 -> java.lang.NullPointerException('null') ***

```
C:\Users\m1srh\Documents\Apeer_Modules\fiji_module_template_b88fae21-a305-4afb-b70b-48c18efc9fa8>docker run -it --rm -v c:\Temp\input:/input -v c:\Temp\output:/output --env-file wfe.env sebi/apeer_test_fijimodule:latest
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option PermSize=128m; support was removed in 8.0
Java HotSpot(TM) 64-Bit Server VM warning: Using incremental CMS is deprecated and will likely be removed in a future release
[INFO] Overriding BIOP Run Macro...; identifier: command:ch.epfl.biop.macrorunner.B_Run_Macro; jar: file:/Fiji.app/plugins/BIOP/B_Run_Macro-1.0.0-SNAPSHOT.jar
[INFO] Overriding Get Spine From Circle Rois; identifier: command:Cirlces_Based_Spine; jar: file:/Fiji.app/plugins/Max_Inscribed_Circles-1.1.0.jar
[INFO] Image Filename : /input/cells.tif
13:19:50.336 [SciJava-563a89b5-Thread-2] ERROR loci.formats.ImageReader -  *** One or more readers is misbehaving. See the debug output for more information. e.g.:
     loci.formats.in.APLReader@44c82053 -> java.lang.NullPointerException('null') ***
[INFO] Apply Filter  : MEDIAN
[INFO] Filter Radius : 5
[INFO] Duration of whole Processing : 2.4412176
[INFO] Saving Filtered Image .../output/cells_FILTERED.ome.tiff
[INFO] Duration of saving as OME.TIFF : 0.5278022
[INFO] Writing output JSON file ...
[INFO] Done.
```

So what does this mean? The output seems OK ....? Here is the script:

```
# @LogService log

# required import
import os
import json
from java.lang import Double, Integer
from ij import IJ, ImagePlus, ImageStack, Prefs
from ij.process import ImageProcessor, LUT
from ij.plugin.filter import RankFilters
from loci.plugins import BF
from loci.plugins.in import ImporterOptions
from loci.plugins import LociExporter
from loci.plugins.out import Exporter
from ij.io import FileSaver
import time

# helper function to apply the filetr
def getImageStack(imp):

    # get the stacks
    try:
        stack = imp.getStack()  # get the stack within the ImagePlus
        nslices = stack.getSize()  # get the number of slices
    except:
        stack = imp.getProcessor()
        nslices = 1

    return stack, nslices

def apply_filter(imp,
                 radius=5,
                 filtertype='MEDIAN'):

    # initialize filter
    filter = RankFilters()

    # create filter dictionary
    filterdict = {}
    filterdict['MEAN'] = RankFilters.MEAN
    filterdict['MIN'] = RankFilters.MIN
    filterdict['MAX'] = RankFilters.MAX
    filterdict['MEDIAN'] = RankFilters.MEDIAN
    filterdict['VARIANCE'] = RankFilters.VARIANCE
    filterdict['OPEN'] = RankFilters.OPEN
    filterdict['DESPECKLE'] = RankFilters.DESPECKLE

    # get the stack and number of slices
    stack, nslices = getImageStack(imp)

    for index in range(1, nslices + 1):
        # get the image processor
        ip = stack.getProcessor(index)
        # apply filter based on filtertype
        filter.rank(ip, radius, filterdict[filtertype])

    return imp

#######################################################################

def run(imagefile, useBF=True, series=0):

    log.info('Image Filename : ' + imagefile)

    if not useBF:
        # using IJ static method
        imp = IJ.openImage(imagefile)

    if useBF:

        # initialize the importer options
        options = ImporterOptions()
        options.setOpenAllSeries(True)
        options.setShowOMEXML(False)
        options.setConcatenate(True)
        options.setAutoscale(True)
        options.setId(imagefile)

        # open the ImgPlus
        imps = BF.openImagePlus(options)
        imp = imps[series]

    # apply the filter
    if FILTERTYPE != 'NONE':

        # apply filter
        log.info('Apply Filter  : ' + FILTERTYPE)
        log.info('Filter Radius : ' + str(FILTER_RADIUS))

        # apply the filter based on the choosen type
        imp = apply_filter(imp,
                           radius=FILTER_RADIUS,
                           filtertype=FILTERTYPE)

    if FILTERTYPE == 'NONE':
        log.info('No filter selected. Do nothing.')

    return imp

##################################################################

# Parse Inputs of Module
INPUT_JSON = json.loads(os.environ['WFE_INPUT_JSON'])
IMAGEPATH = INPUT_JSON['IMAGEPATH']

# suffix for the filename of the saved data
SUFFIX_FL = '_FILTERED'

# parameters for filter
FILTERTYPE = INPUT_JSON['FILTERTYPE']
FILTER_RADIUS = INPUT_JSON['FILTER_RADIUS']
SAVEFORMAT = 'ome.tiff'

# define path for the output
outputimagepath = '/output/' + os.path.basename(IMAGEPATH)
basename = os.path.splitext(outputimagepath)[0]

# remove the extra .ome before reassembling the filename
if basename[-4:] == '.ome':
    basename = basename[:-4]
    log.info('New basename for output :' + basename)

# save processed image
outputimagepath = basename + SUFFIX_FL + '.' + SAVEFORMAT

#############   RUN MAIN IMAGE ANALYSIS PIPELINE ##########

# get the starting time of processing pipeline
start = time.clock()

# run image analysis pipeline
filtered_image = run(IMAGEPATH,
                     useBF=False,
                     series=0)

# get time at the end and calc duration of processing
end = time.clock()
log.info('Duration of whole Processing : ' + str(end - start))

###########################################################

# save the particle stack
outputimagepath = '/output/' + os.path.basename(IMAGEPATH)

# save processed image in out_dir
outputimagepath = os.path.splitext(outputimagepath)[0] + SUFFIX_FL + '.' + SAVEFORMAT
log.info('Saving Filtered Image ...' + outputimagepath)

# create the argument string for the BioFormats Exporter and save as OME.TIFF
start = time.clock()
paramstring = "outfile=" + outputimagepath + " " + "windowless=true compression=Uncompressed saveROI=false"
plugin = LociExporter()
plugin.arg = paramstring
exporter = Exporter(plugin, filtered_image)
exporter.run()
# get time at the end and calc duration of processing
end = time.clock()
log.info('Duration of saving as OME.TIFF : ' + str(end - start))

# write output JSON
log.info('Writing output JSON file ...')
output_json = {"FILTERED_IMAGE": outputimagepath}

with open("/output/" + INPUT_JSON['WFE_output_params_file'], 'w') as f:
    json.dump(output_json, f)

# finish
log.info('Done.')

# finish and exit the script
os._exit()
```

joshmoore, 2019-05-22:
Hi @sebi06. The error message was added in https://github.com/openmicroscopy/bioformats/pull/3344 to allow execution to continue despite the NullPointerException. You can read more about the root cause in https://github.com/ome/ome-common-java/pull/38 -- but the full solution required more changes than we wanted to attempt for 6.1.0. A workaround to remove the error message is to move the files as @dgault explains in https://forum.image.sc/t/bioformats-importer-issue/25439/2?u=joshmoore

Hope that helps.
~Josh

sebi06, 2019-05-22:
> about the root cause in ome/ome-common-java#38

Thx for the answer. This helps and I can live with this for now easily.
The error just popped up when running the script inside a docker on APEER
inside the logs.