ome/bioformats #3381: opening zeiss czi fastairyscan images

sdutertre, 2019-05-27:
Hello,

I'm trying to open fastairyscan raw data in FIJI but all the pixels of the images are at O except in channel 2 (sum of the images). Images from conventional airyscan open correctly. Any suggestion/ideas are welcome.
thanks
Stephanie

mtbc, 2019-05-28:
Sorry to hear you've run into an image reading bug. I don't think we've seen this one before. Could you share one of your images with us so we can look into what is going wrong? You can send via http://qa.openmicroscopy.org.uk/qa/upload/ (mention this issue) or make one available for us to download, whichever is easiest.

sdutertre, 2019-05-28:
Thank you for your quick reply. Here is the link to download an image: 
https://filesender.renater.fr/?s=download&token=f45c9006-e32a-7d9f-ca26-c362ccad6848

mtbc, 2019-05-28:
Great, thank you, we now have it safely in hand.

dgault, 2019-05-29:
Hi @sdutertre, I have been testing the submitted image today and can reproduce the issues you are seeing using the latest Bio-Formats release. Do the dimensions of the file or any of the metadata look incorrect or is it purely the 0 pixel values?

sdutertre, 2019-06-04:
Hi David,
For me, the metadata seems correct. The problem is that we have these 0 pixel images instead of our data.