ome/bioformats #3389: Error in bfGetPlane when loading .tif file

will-h-uon, 2019-06-12:
I've been using bfopen to load my tif files without issue, however this file gives the following error in MATLAB:

```
Imstack = bfopen('hela_ctrl_brown_exp_1_cell_3_tr_70_MMStack.ome.tif');
Reading series #1
    .Error using bfGetPlane (line 78)
Java exception occurred:
java.lang.NullPointerException
        at loci.formats.tiff.TiffParser.getTile(TiffParser.java:690)
        at loci.formats.tiff.TiffParser.getSamples(TiffParser.java:982)
        at loci.formats.tiff.TiffParser.getSamples(TiffParser.java:779)
        at loci.formats.in.MinimalTiffReader.openBytes(MinimalTiffReader.java:303)
        at loci.formats.DelegateReader.openBytes(DelegateReader.java:227)
        at loci.formats.ImageReader.openBytes(ImageReader.java:457)
        at loci.formats.ChannelFiller.openBytes(ChannelFiller.java:156)
        at loci.formats.ChannelSeparator.openBytes(ChannelSeparator.java:227)
        at loci.formats.ChannelSeparator.openBytes(ChannelSeparator.java:159)

Error in bfopen (line 150)
        arr = bfGetPlane(r, i, varargin{:});
```

The stack was supposed to have 20,000 T positions, but when I've loaded it in FIJI, it only has ~3,000 - could the problem be because it's an incomplete file? Is there a way to check?

dgault, 2019-06-13:
Hi @will-h-uon, if you wish we can test the sample file for you if you upload it to https://www.openmicroscopy.org/qa2/qa/upload/. What software are you using to acquire the images?

will-h-uon, 2019-06-14:
Hi, I sent a reply by email about the file - I was unable to upload it with your uploader because the flash applet didn't run (I'm using Ubuntu 14.04, with up-to-date flash player).

I acquired the data using micro-manager.

dgault, 2019-06-18:
Hi @will-h-uon, who did you send the email to?

will-h-uon, 2019-06-18:
The email from Github said "you can reply to this email directly", which I hoped would mean you get my reply. I've copied my email below:

"Hi, I'm unable to use the file uploader you linked - I get an error:

"SWFUpload is taking a long time to load or the load has failed. Please make sure that the Flash Plugin is enabled and that a working version of the Adobe Flash Player is installed. "

I updated flash, but I think it's because I'm using an Ubuntu 14.04 which is a few years old. I've uploaded the file to [hela_ctrl_brown_exp_1_cell_3_tr_70_MMStack.ome.tif](https://uniofnottm-my.sharepoint.com/:i:/g/personal/william_hardiman_nottingham_ac_uk/EexmKUUhkV5GpJ0mHAywjzUB5X7gO2DN8sFmMLf12A6_Fw?e=RScamP), although I appreciate that it's not going through the normal system so I hope that doesn't make this difficult for you.

Thanks,

Will"

dgault, 2019-06-19:
Thanks @will-h-uon, I have been able to test the file today and reproduce the issue. The issue seems to be with the very first plane which does not have any offset values. I can put in place a fix that would prevent the error and allow the image to be opened but it would display an empty plane for the first plane, the remaining 3118 planes all appear to be correct. 

Was the image acquired in MicroManager? Would you expect the first plane to differ in any way?

will-h-uon, 2019-06-20:
The image was acquired in MicroManager, yes. I can't think of any reason why it would differ.