ome/bioformats #3397: CZI from AxioScan does not open correctly with BioFormtas 6.1.1

sebi06, 2019-06-28:
Hi,

I have a CZI that does open correctly with BioFormats 6.1.1. I did the following

- open compressed CZI in Fiji
- open uncompressed CZI in Fiji
- export CZI as OME-TIFF from ZEN and open in Fiji
- export CZI as BigTIFF from ZEN and open in Fiji

None of those approached works correctly. But when I open the BigTIFF inside another (non-BioFormats based viewer) it open correctly without artifacts, which makes we wonder, if this a bug in BioFormats.

At the very bottom I also attached a screenshot of ZEN and the llink to the CZI:

https://www.dropbox.com/s/5guj227wt7spk1l/2019_06_21__13_01__0095.czi?dl=0

[image]

[image]

mtbc, 2019-07-02:
Thank you, sample files and corresponding screenshots are always helpful. I've now placed `2019_06_21__13_01__0095.czi` where the Bio-Formats team can experiment with it.

sebi06, 2019-07-02:
No problem. Let me know if your need more info. We really need to sort this out since the user (are understandably) quite unhappy with this ....

dgault, 2019-07-05:
Hi @sebi06, I have been testing the submitted file today and so far I've not been able to reproduce the issue. Do you still see the issue if you use the `showinf` command line tool? (https://www.openmicroscopy.org/bio-formats/downloads/, docs on the showinf command - https://docs.openmicroscopy.org/bio-formats/6.1.1/users/comlinetools/display.html)

Im wondering if it is perhaps related to https://forum.image.sc/t/polygon-tool-distorts-series-read-into-fiji-via-the-bio-formats-importer-plugin/27249

sebi06, 2019-07-06:
I just tried it out at home on Ubuntu and got stuck here:

(base) sebi06@sebastian-tuxedo:/datadisk1/tuxedo/BioFormats/6.1.1$ java -XshowSettings:vm
VM settings:
    Max. Heap Size (Estimated): 7.79G
    Using VM: Java HotSpot(TM) 64-Bit Server VM

So i figured this should be OK for an 180MB CZI file, but it turns out it is not ... But I do not really know yet, how to adress this issue.

(base) sebi06@sebastian-tuxedo:/datadisk1/tuxedo/BioFormats/6.1.1$ bash showinf /datadisk1/Sebastian/Dropbox/CZI_Testdata/2019_06_21__13_01__0095.czi
Checking file format [Zeiss CZI]
Initializing reader
ZeissCZIReader initializing /datadisk1/Sebastian/Dropbox/CZI_Testdata/2019_06_21__13_01__0095.czi
Initialization took 1.935s

Reading core metadata
filename = /datadisk1/Sebastian/Dropbox/CZI_Testdata/2019_06_21__13_01__0095.czi
Series count = 8
Series #0 :
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 18935
	Height = 30364
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 79 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #1 :
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 9467
	Height = 15182
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 79 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #2 :
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 4733
	Height = 7591
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 79 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #3 :
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 2366
	Height = 3795
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 79 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #4 :
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 1183
	Height = 1897
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 79 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #5 :
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 591
	Height = 948
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 591 x 948
	Thumbnail size = 79 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #6 :
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 518
	Height = 668
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 518 x 337
	Thumbnail size = 99 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint16
	Valid bits per pixel = 12
	Metadata complete = false
	Thumbnail series = true
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #7 :
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (false color)
	Width = 1500
	Height = 723
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1500 x 116
	Thumbnail size = 128 x 61
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint16
	Valid bits per pixel = 12
	Metadata complete = false
	Thumbnail series = true
	-----
	Plane #0 <=> Z 0, C 0, T 0

Reading series #0 pixel data (0-0)
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
	at loci.common.DataTools.allocate(DataTools.java:1260)
	at loci.formats.FormatReader.openBytes(FormatReader.java:869)
	at loci.formats.ImageReader.openBytes(ImageReader.java:445)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
	at loci.formats.gui.BufferedImageReader.openImage(BufferedImageReader.java:86)
	at loci.formats.tools.ImageInfo.readPixels(ImageInfo.java:821)
	at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1055)
	at loci.formats.tools.ImageInfo.main(ImageInfo.java:1121)
(base) sebi06@sebastian-tuxedo:/datadisk1/tuxedo/BioFormats/6.1.1$

sebi06, 2019-07-06:
And when I try to open the CZI on the same system with Fiji I get no java heap spcae issues. The image at first glance looks also distorted.
But when I then run split channels on the image with the aritfacts the resulting 3 channel look OK - ???

[image]