ome/bioformats #3399: Can't load ome-tiff subIFDs with a value greater than‭ 42949672954

jordigarnacho, 2019-07-01:
Hi,

I can not open the images with pyramids having a subifd value written on 64 bits (long). I have no problem in case I use images smaller than 4GB (with subids less than 42949672954).

> java.lang.IllegalArgumentException: Invalid array size: 0 x 0
> 	at loci.common.DataTools.safeMultiply32(DataTools.java:1281)
> 	at loci.formats.in.OMETiffReader.addSubResolutions(OMETiffReader.java:1335)
> 	at loci.formats.in.OMETiffReader.initFile(OMETiffReader.java:1227)
> 	at loci.formats.FormatReader.setId(FormatReader.java:1395)
> 	at loci.formats.ImageReader.setId(ImageReader.java:842)
> 	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
> 	at loci.formats.Memoizer.setId(Memoizer.java:662)

Thanks.

joshmoore, 2019-07-01:
What version of Bio-Formats are using to read the files and how were the files produced?

jordigarnacho, 2019-07-01:
Thanks for your reply. 
I use QuPath 0.2.0 m2 with Bio-Format 6.0.1. 
To generate the images I use a personal algorithm in java.

joshmoore, 2019-07-01:
You might try Bio-Formats 6.1.1. If you are still having issues, can you share a file and optimally the code used to generate the file? Thanks. ~J.

jordigarnacho, 2019-07-01:
The images are available at: [https://send.firefox.com/download/cd5fa10e9fd5f32a/#Di4Oldwsi0U7RgqQwyjW4A](https://send.firefox.com/download/cd5fa10e9fd5f32a/#Di4Oldwsi0U7RgqQwyjW4A).
I zipped the files to reduce the size.

jordigarnacho, 2019-07-02:
I tried with Bio-Format 6.1.1. I have the same problem. Do you need more information? Thanks for your help.

dgault, 2019-07-02:
Hi @jordigarnacho, Im having trouble decompressing the files, can you upload them to https://www.openmicroscopy.org/qa2/qa/upload/?

From the exception you are seeing this appears to be caused by the tile sizes. Do you know the image and dimensions (X and Y) for the problem file?

jordigarnacho, 2019-07-02:
Hi @dgault, I just uploaded the files. 
- subResolutionsValid.zip should be decompressed into subResolutionsValid.ome.tf2. This is the image of an image that I can open.
- subResolutionsError.zip should be decompressed into subResolutionsError.ome.tf2. This is the image of an image with which I have the error.

Valid image properties : 
- Image Width : 20000px, Image Length : 19500px,Tile Width : 256px, Tile Height 256px.

Error image properties : 
- Image Width : 20000px, Image Length : 20000px,Tile Width : 256px, Tile Height 256px.

Both images were created with the same algorithm. I only change the dimension of the image.

dgault, 2019-07-03:
Thanks @jordigarnacho, from parsing through the 2 files, the first has 7 valid subIFD's for the sub resolutions. For the file that is erroing, there are 7 subIFD's but only the first 2 appear to be populated correctly:

The first contains the below which looks correct:
`{0=true, 256=10000, 1=true, 257=10000, 258=16, 322=256, 259=1, 323=256, 324=[J@431ae79b, 325=[J@42cba53, 262=1, 339=1, 277=1, 282=2500/1, 283=2500/1, 284=1, 255=2}`

The second also looks correct:
`{0=true, 256=5000, 1=true, 257=5000, 258=16, 322=256, 259=1, 323=256, 324=[J@6f3f6304, 325=[J@78053343, 262=1, 339=1, 277=1, 282=1250/1, 283=1250/1, 284=1, 255=2}`

All the rest contain:
`{0=true, 1=true}`

cgohlke, 2019-07-03:
> The images are available at: https://send.firefox.com/download/cd5fa10e9fd5f32a/#Di4Oldwsi0U7RgqQwyjW4A.

This link has expired.

jordigarnacho, 2019-07-08:
@dgault Thanks a lot for your answer. Indeed, there is a problem writing IFDs in my image generation algorithm. I will correct that.
Thank you so much for your help.

closed 2019-07-08 by jordigarnacho