ome/bioformats #3409: bfsave fails in Octave

carandraug, 2019-07-20:
Reported on the [image.sc forum](https://forum.image.sc/t/bio-formats-octave-error-on-bfsave/27504), basically `bfsave` fails in Octave:

```
bfsave(zeros(100,100,2,1,3), 'test.tiff')
error: [java] java.lang.NoSuchMethodException: setPixelsBigEndian
error: called from
    createMinimalOMEXMLMetadata at line 61 column 1
    bfsave at line 78 column 14
```

I believe this is a recent bug in Octave, related to the rules of auto boxing/unboxing (or packing/unpacking) between java and Octave types, but I'm not sure exactly where.

carandraug, 2019-07-20:
The fix proposed on that forum works, and can be pulled from my `octave-java_get-boolean` branch. But this is a workaround for a bug in Octave.

sbesson, 2019-07-21:
Hi @carandraug

See #3405 for the open PR containing the workaround proposed in the image.sc
thread.

Since you implemented the Bio-Formats Octave package I wanted to confirm
this was acceptable to you and likely to work across multiple versions of
GNU Octave.

carandraug, 2019-07-22:
Oh, I missed that PR because I searched on issues and not in pulls. I've left comments there now. Closing.

closed 2019-07-22 by carandraug