ome/bioformats #3438: Failing to load VSI grouped files

pbl007, 2019-09-13:
An apparent tag reading issue halts the operation to read a group of CellSens (.vsi) files. The following is from the debug window:

out>   dataFieldOffset = 5914846652683978913
 
out>   flags = 858058961
 
out> Failed to read all tags
java.lang.IllegalArgumentException: null
	at java.nio.Buffer.position(Buffer.java:244)
	at loci.common.NIOFileHandle.buffer(NIOFileHandle.java:650)
	at loci.common.NIOFileHandle.seek(NIOFileHandle.java:330)
	at loci.common.RandomAccessInputStream.seek(RandomAccessInputStream.java:203)
	at loci.formats.in.CellSensReader.readTags(CellSensReader.java:1888)
	at loci.formats.in.CellSensReader.initFile(CellSensReader.java:647)
	at loci.formats.FormatReader.setId(FormatReader.java:1389)
	at loci.formats.ImageReader.setId(ImageReader.java:843)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.DimensionSwapper.setId(DimensionSwapper.java:293)
	at loci.formats.FileStitcher.initFile(FileStitcher.java:959)
	at loci.formats.FileStitcher.setId(FileStitcher.java:925)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelFiller.setId(ChannelFiller.java:223)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelSeparator.setId(ChannelSeparator.java:293)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.DimensionSwapper.setId(DimensionSwapper.java:293)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.plugins.in.ImportProcess.setId(ImportProcess.java:701)
	at loci.plugins.in.ImportProcess.initializeStack(ImportProcess.java:547)
	at loci.plugins.in.ImportProcess.execute(ImportProcess.java:146)
	at loci.plugins.in.Importer.showDialogs(Importer.java:140)
	at loci.plugins.in.Importer.run(Importer.java:76)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:230)
	at ij.IJ.runPlugIn(IJ.java:193)
	at ij.IJ.runPlugIn(IJ.java:182)
	at loci.plugins.shortcut.ShortcutPanel$1.run(ShortcutPanel.java:189)

mtbc, 2019-09-13:
Initial discussion occurring on https://forum.image.sc/t/29456.

dgault, 2021-05-27:
The PR to resolve this issue has been tested and merged and will be available in the upcoming Bio-Formats 6.7.0. Closing this issue.

PR #3690 (2021-05-27): CellSens vsi: fix offset check to prevent seeking beyond EOF
Fixes #3438

To test, use ```showinf -nopix -debug``` on .vsi files in ```inbox/gh-3438```.  Without this PR, an exception should be logged as noted in #3438.  With this PR, there should be no exception.

I would not expect any test or memo file impact; this should be safe for a patch release.
commit 302d9088 (2021-05-27): CellSens vsi: fix offset check to prevent seeking beyond EOF

Fixes #3438
closed 2021-05-27 by dgault