ome/bioformats #3440: Exposing original image tile bytes

ebremer, 2019-09-18:
I've been converting a copius number of VSI images to bigtiff as our toolings won't work with VSI.  I know bioformats insists on decoding and re-encoding image tiles when converting but since no loss of quality is of paramount importance, we use LZW compression which has the undesired affect of ballooning the image to almost 10x it's size.  I was going to use JPG as a comparison to see if the image quality changed dramatically which spawned https://github.com/ome/bioformats/issues/3370.
I tried an experiment and modified CellSenseReader.java to expose the original tile bytes.  I was able to retrieve the VSI tiles and route them into a file tile.jpg.  If the tile size is kept the same, it doesn't seem that there is any reason why the decoding/re-encoding process needs to occur making for a faster conversion from VSI->TIFF and possible other image formats. (unless I am missing something)

So here is the big ask, would it be difficult to augment the general Bioformats API to expose the original tile bytes and then just route them pass the decode/encode process of image conversion providing the tile size is not changed into the new tile "framing", tiff in this case?

melissalinkert, 2023-06-16:
#3992 is some initial work on this topic (though not implemented for VSI at the moment).

ebremer, 2024-03-11:
Should this work with all readers?

melissalinkert, 2024-03-12:
> Which readers does -precompressed work with?

Only SVS at the moment. The API is in place for other readers to extend, but we have not yet expanded this feature to additional readers.

ebremer, 2024-05-09:
Any timeline for when other formats will be supported for precompressed  to work with?

sbesson, 2025-06-09:
As of Bio-Formats 8.2.0 a few readers support reading precompressed tiles from the original format and a few writers support writing precompressed tiles - see https://bio-formats.readthedocs.io/en/latest/formats/precompressed.html for the up-to-date list. Closing this issue as the API is in-place and additional readers can be discussed in separate issues