ome/bioformats #3449: Point Names missing in ND2 Metadata

kevinjohncutler, 2019-10-08:
I cannot find the point names in the metadata for images captured in the nd2 format via NIS Elements. The xy positions can be labeled (called "point names"),and they are loaded when the nd2 file is loaded back into Elements. The metadata collected with bfGetReader seems not to find this. Note that the NIS Viewer (on macOS, at least) does not display this information either. It must be hidden somewhere, however, if Elements is able to find it. 

This info is grouped with the multipoint acquisition settings that get loaded into Elements. so maybe there is a way to get to that info?

dgault, 2019-11-25:
Hi @kevinjohncutler, apologies for the late reply, do you have a sample file you would be able to upload to https://www.openmicroscopy.org/qa2/qa/upload/

kevinjohncutler, 2019-12-06:
@dgault no worries, thanks for checking into it! I am uploading 'null_vs_bind_0-5-10_003.nd2'. The names are null_y_0_a, bind_y_10_e, etc.