ome/bioformats #3450: shortLut causes NullPointException

ltwonders, 2019-10-10:
When parsing a dicom file with "Red Palette Color LUT Data" key, NullPointException occurs as below:
```java
java.lang.NullPointerException
	at loci.formats.in.DicomReader.addInfo(DicomReader.java:953)
	at loci.formats.in.DicomReader.initFile(DicomReader.java:685)
	at loci.formats.FormatReader.setId(FormatReader.java:1395)
	at loci.formats.ImageReader.setId(ImageReader.java:842)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelSeparator.setId(ChannelSeparator.java:291)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
```
This exception occurs in both 6.0.0-m2 and 6.2.1 as I tried. It seems the `this.shortLut` and `this.lut` are not initialized before assign  values.

sbesson, 2019-10-30:
Hi @ltwonders, sorry for the delay here. This issue has been missed from our radar. Do you have a sample file allowing to reproduce this error? If small enough and public, you should be able to attach it to this issue. Otherwise, you can upload one at http://qa.openmicroscopy.org.uk/qa/upload/

ltwonders, 2019-10-31:
Sorry that I can't upload the pic as privacy. However after I debug this issue, I resolve it with some code adding to the `DicomReader.addInfo()` method as:
```java
if(null == this.shortLut) {
    this.shortLut = new short[3][];
}
this.shortLut[ndx] = new short[this.elementLength / 2];
if(null == this.lut){
    this.lut = new byte[3][];
}
this.lut[ndx] = new byte[this.elementLength / 2];
```
The pic has key `Palette Color LUT Data` leads to this issue.

sbesson, 2019-10-31:
Ah thanks for the extra code and the debugging, the `lut` and shortLut` seem to be initialized  during the processing of the `Photometric Interpretation` tag.

https://github.com/ome/bioformats/blob/584ab98f489398e16e7ce8cdd7a5c80dab133cde/components/formats-bsd/src/loci/formats/in/DicomReader.java#L926-L937

In your sample file is the `Photometric Interpretation` tag  present and set to `Palette Color`?

ltwonders, 2019-11-01:
After I try `pydicom` to read the file, some dataset that maybe helpful shows as below :
```
(0028, 0004) Photometric Interpretation          CS: 'MONOCHROME2'
(0028, 1101) Red Palette Color Lookup Table Desc SS: [2965, 45, 8]
(0028, 1102) Green Palette Color Lookup Table De SS: [2965, 45, 8]
(0028, 1103) Blue Palette Color Lookup Table Des SS: [2965, 45, 8]
(0028, 1201) Red Palette Color Lookup Table Data OW: Array of 2966 elements
(0028, 1202) Green Palette Color Lookup Table Da OW: Array of 2966 elements
(0028, 1203) Blue Palette Color Lookup Table Dat OW: Array of 2966 elements
(0008, 0070) Manufacturer                        LO: 'GE MEDICAL SYSTEMS'
(0008, 1090) Manufacturer's Model Name           LO: 'DISCOVERY MR750'
```
BTW,  `pydicom` can read this file well.

sbesson, 2019-11-08:
Thanks for the additional information. We are not experts of the DICOM standard but I suspect the issue comes from an assumption in the reader that LUT should only be present if the `Photometric Interpretation` is set to `Palette Color` as per http://dicom.nema.org/medical/dicom/current/output/chtml/part03/sect_C.7.6.3.html#sect_C.7.6.3.1.2.

Improving the robustness of the code to handle files like yours makes sense.  even though I am not sure  how a monochrome image plane with RGB lookup tables should be interpreted. In all cases, feel free to open a PR with your proposed diff. In the absence of a sample file allowing us to reproduce the problem, we can minimally make sure it does not cause any regression against our sample DICOM files.

melissalinkert, 2023-09-14:
Closing, as I'm fairly certain this would have been fixed by the DICOM reader reworking in #3742; that was included in 6.8.0 and later. @ltwonders, feel free re-open if you still see an exception with a current version of Bio-Formats.

closed 2023-09-14 by melissalinkert